dji-sdk/Onboard-SDK #624: cannot compile OSDK with clang++ 10.0.0 in macOS 10.15.6

lbdhr, 2020-09-03:
I am using the macOS 10.15.6. The version of clang and clang++ is `Apple clang version 11.0.3 (clang-1103.0.32.62)`. The version of make is `GNU Make 4.3 Built for x86_64-apple-darwin19.2.0`. The version of cmake is `cmake version 3.18.2`. The version of OSDK is 3.8.

When I typed `cmake ..` in the build directory, everything works fine. The output is:
```
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling without WaypointV2 Core
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/build
```

However, when I typed the 'make ' command, it showed as the following:
```
Scanning dependencies of target djiosdk-core
[  3%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_ack.cpp.o
[  6%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_broadcast.cpp.o
In file included from /Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/src/dji_broadcast.cpp:31:
In file included from /Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_vehicle.hpp:42:
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:98:5: error: unknown type name 'RecvTimeMsg'
    RecvTimeMsg timestamp; // this is OSDK recv time
    ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:219:20: error: unknown type name 'HWSyncDataFlag'
  void setDataFlag(HWSyncDataFlag &flag, bool val);
                   ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:225:20: error: unknown type name 'HWSyncDataFlag'
  bool getDataFlag(HWSyncDataFlag &flag);
                   ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:235:26: error: unknown type name 'RecvTimeMsg'
  void recordRecvTimeMsg(RecvTimeMsg &recvTime);
                         ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:237:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag GPGSAFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:238:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag GLGSAFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:239:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag GAGSAFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:240:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag BDGSAFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:241:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag GPRMCFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:242:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag UTCFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:243:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag fcTimeFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:244:3: error: unknown type name 'HWSyncDataFlag'
  HWSyncDataFlag ppsSourceFlag;
  ^
/Users/libai/Desktop/workspace_cloud/UAV/Onboard-SDK-3.8/osdk-core/api/inc/dji_hardware_sync.hpp:247:24: error: unknown type name 'HWSyncDataFlag'
  bool writeDataHelper(HWSyncDataFlag &flag,
                       ^
13 errors generated.
gmake[2]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/build.make:95: osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_broadcast.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:113: osdk-core/CMakeFiles/djiosdk-core.dir/all] Error 2
gmake: *** [Makefile:149: all] Error 2
```

Could you please help me? Thank you very much.

dji-dev, 2020-09-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38807'>Zendesk ticket #38807</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

We are sorry that currently OSDK is not compatible with MacOS, This may have some compatibility problems.

Considering the practical application, and it may not be common to use MACOS as an onboard computer platform in practical applications. It is recommended to use the official recommended platform Ubuntu1604 and Using a virtual machine is also optional.

 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/iQm9kSiNDs7qkNVjtFlJrt3hG' target='zd_attach'>inline2054195162.png</a>

lbdhr, 2020-09-04:
Thank you very much for the quick reply. I will continue to use the ubuntu1604 in Parallels.

closed 2020-09-04 by lbdhr