dji-sdk/Onboard-SDK #625: M300 waypoint v2 mission problem with BR v4.0.1

XiaoR, 2020-09-04:
Hi,
I just update the osdk to git tag v4.0.1, but when running the waypoint v2 sample , I got the following unknown error code:

# ./djiosdk-waypoint-v2-sample 
Read App ID
User Configuration read successfully. 

[1493889.081]STATUS/1 @ parseDroneVersionInfo, L1157: Device Serial No. = 1ZNDH7800A14XK
[1493889.081]STATUS/1 @ parseDroneVersionInfo, L1159: Firmware = 3.4.8.44
[1493889.081]STATUS/1 @ functionalSetUp, L315: Shake hand with drone successfully by getting drone version.
[1493889.082]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[1493890.082]STATUS/1 @ sendHeartbeatToFCTask, L1483: OSDK send heart beat to fc task created.
[1493890.282]STATUS/1 @ FileMgrImpl, L234: register download file callback handler successfully.
[1493890.282]STATUS/1 @ MopPipelineManagerBase, L48: MOP background task now is created .
[1493891.306]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[1493891.306]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[1493891.306]STATUS/1 @ RegisterOSDInfoCallback, L535: register result of geting　FC ground station status pushing : 1

[1493891.306]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[1493891.335]STATUS/1 @ init, L255: Start bulk channel for M300's liveview!
[1493891.335]STATUS/1 @ init, L262: Create task for M300's liveview!
[1493891.337]STATUS/1 @ init, L270: Start bulk channel for M300's perception
[1493891.338]STATUS/1 @ init, L277: Create task for M300's advanced sensing!
[1493891.377]STATUS/1 @ parseDroneVersionInfo, L1157: Device Serial No. = 1ZNDH7800A14XK
[1493891.377]STATUS/1 @ parseDroneVersionInfo, L1159: Firmware = 3.4.8.44
[1493891.378]STATUS/1 @ AdvancedSensing, L104: Advanced Sensing init for the M300 drone
[1493891.678]STATUS/1 @ init, L283: Start advanced sensing initalization
[1493891.678]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[1493891.678]STATUS/1 @ activate, L1248: osdk policy file updating(1) ......
[1493892.680]STATUS/1 @ activate, L1256: osdk policy file updating(2) ......
[1493892.688]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[1493892.688]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[1493892.691]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[1493892.692]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[1493892.694]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[1493892.694]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 160
[1493892.698]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[1493892.698]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[1493893.680]STATUS/1 @ activate, L1268: version 0x304082C

[1493893.695]STATUS/1 @ activate, L1292: Activation successful

[1493894.736]STATUS/1 @ verify, L244: Verify subscription successful.
[1493894.773]STATUS/1 @ startPackage, L345: Start package 0 result: 0.
[1493894.773]STATUS/1 @ startPackage, L347: Package 0 info: freq=1, nTopics=1.

[1493894.778]STATUS/1 @ runWaypointV2Mission, L116: Set up subscription successfully!

[1493895.788]STATUS/1 @ initMissionSetting, L220: Init mission setting successfully!

[1493896.799]STATUS/1 @ uploadWaypointMission, L237: Upload waypoint v2 mission successfully!

[1493897.806]STATUS/1 @ downloadWaypointMission, L253: Download waypoint v2 mission successfully!

[1493900.077]ERRORLOG/1 @ uploadWapointActions, L263: Upload waypoint v2 actions ErrorCode:0x3

[1493900.077]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[1493900.077]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Request time out
[1493900.077]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Try again or check the status of the target object.
[1493901.080]ERRORLOG/1 @ startWaypointMission, L278: Start waypoint v2 mission ErrorCode:0x110000010040

[1493901.081]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : WaypointV2
[1493901.081]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Unknown
[1493901.081]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution :

dji-dev, 2020-09-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38835'>Zendesk ticket #38835</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Could you please provide the firmware version of M300 , and suggest that you update the latest firmware(01.00.02.11), and then test it with OSDK v4.0.1.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

XiaoR, 2020-09-23:
Confirmed that  waypoint v2 mission works on OSDK v4.0.1 with M300's firmware(01.00.02.11),  but there's something wrong with the coordinates, the drone cannot fly to the exact locations of the waypoints I uploaded.

dji-kyle, 2021-06-24:
update to the latest firmware and OSDK version

closed 2021-06-24 by dji-kyle