dji-sdk/Onboard-SDK #627: M600 SDK questions: Version and telemetry data

tkazik, 2020-09-07:
Dear Dev Team

We have some questions regarding the M600 SDK:
- Which OSDK version would you **recommend**? The latest release is v3.9, dated 2019-09-23.
- However, there is a this [newer commit](https://github.com/dji-sdk/Onboard-SDK/commit/0129954701a8e659659f8993940d1171aef7a20f) (dated 2019-12-03) that contains many updates. Is it an unofficial v3.9.1 or does 'for review' imply 'for documentation' only.

We are interested in using the telemetry data, especially the data from the ESC:
- OSDK v3.7 (dated 2018-08-05) introduced many new telemetry data including ESC. However, [this line](https://github.com/dji-sdk/Onboard-SDK/blob/3.7/osdk-core/api/inc/dji_telemetry_doc.hpp#L617) mentions that for the M600, this will only become available with an updated firmware.
- In the meantime, the firmware update V01.00.01.67 was released on 2019-02-26. According to the [release notes](https://dl.djicdn.com/downloads/m600/20190226/M600_Release_Notes_en_20190226.pdf), it introduces stopping the motors in-flight (we need this feature as well), but noting is mentioned about the telemetry.
- In the newest version of the SDK (v3.9 or v3.9.1?), it still says that ESC data will be available with an updated firmware for the M600, see [this line](https://github.com/dji-sdk/Onboard-SDK/blob/3.9/osdk-core/api/inc/dji_telemetry_doc.hpp#L615).

So there are two options:
- Either the firmware update did not introduce the ESC data feature for the M600.
- Or, the comment in the code of v3.9 was not updated and ESC data is now actually supported with the latest firmware.

Can you please clarify? Thank you very much!

dji-dev, 2020-09-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38892'>Zendesk ticket #38892</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、Which OSDK version would you **recommend**? The latest release is v3.9, dated 2019-09-23.
     --> For M600 the OSDK 3.9 is recommended ,
 [image]​

2、The drone M600 with latest firmware :V01.00.01.67 not support the TOPIC: TOPIC_ESC_DATA  to obtain ESC data,
And so sorry for the  unclear description.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/Q4JZyD4zXjWaznios8rqdyz5e' target='zd_attach'>inline-1278723215.png</a>

tkazik, 2020-09-08:
Thank you for the quick update.
It is a pity that apparently quite a few topics are not available in the M600.
A clearer documentation would definitively help.
Also linking to this related [issue](https://github.com/dji-sdk/Onboard-SDK/issues/392#issuecomment-459468363), hoping it will help someone in the future.

closed 2020-09-08 by tkazik