dji-sdk/Onboard-SDK #628: H20T advanced sensing LIVEVIEW

IuliuNovac, 2020-09-07:
Hi,

How can I read the stream from two different h264 sources using LiveView simultaneously?
OSDK_CAMERA_SOURCE_H20T_IR and OSDK_CAMERA_SOURCE_H20T_ZOOM

Best regards

dji-dev, 2020-09-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38906'>Zendesk ticket #38906</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry the OSDK not support to get two different stream simultaneously from the same Camera(H20) for OSDK_CAMERA_SOURCE_H20T_IR and OSDK_CAMERA_SOURCE_H20T_ZOOM.

If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

tuyaliang, 2020-09-09:
> Agent comment from kyle.cai in [Zendesk ticket #38906](https://djisdksupport.zendesk.com/agent/tickets/38906):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> Sorry the OSDK not support to get two different stream simultaneously from the same Camera(H20) for OSDK_CAMERA_SOURCE_H20T_IR and OSDK_CAMERA_SOURCE_H20T_ZOOM.
> 
> If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information, https://formcrafts.com/a/dji-developer-feedback-en
> In order to better match your requirements, try to include the following information when filling in：
> 
>     1. Desired functions and requirements
> 
>     2. Company/organization profile
> 
>     3. Project descriptions that use this feature requirement.
> 
>     4. Personal background or brief introduction.
> 
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team

We also need the function of getting two different stream simultaneously from the same Camera(H20) for OSDK_CAMERA_SOURCE_H20T_IR and OSDK_CAMERA_SOURCE_H20T_ZOOM, Is there a development plan？

IuliuNovac, 2020-09-09:
> We also need the function of getting two different stream simultaneously from the same Camera(H20) for OSDK_CAMERA_SOURCE_H20T_IR and OSDK_CAMERA_SOURCE_H20T_ZOOM, Is there a development plan？

It's pretty much essential for osdk, i am surprised it wasn't developed.

IuliuNovac, 2020-09-09:
> We also need the function of getting two different stream simultaneously from the same Camera(H20) for OSDK_CAMERA_SOURCE_H20T_IR and OSDK_CAMERA_SOURCE_H20T_ZOOM, Is there a development plan？

I am pretty sure there is quite a bit of interest in this function. Anyone else, please leave a comment if it will benefit your product.
I already opened a ticket, but feel free to open an another one. Since that should make it a higher priority. 

If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information, https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：

1.     Desired functions and requirements
2.     Company/organization profile
3.     Project descriptions that use this feature requirement.
4.     Personal background or brief introduction.

dji-kyle, 2021-06-24:
Not support to get two different stream simultaneously from the same Camera(H20)

closed 2021-06-24 by dji-kyle