dji-sdk/Onboard-SDK #633: Segmentation fault running in container

arosendorff, 2020-09-15:
Hi,

I'm currently trying to run the OSDK on a docker container, but I seem to be having issues with some linker code giving me segmentation faults (seems to be all the `OsdkOsal_` methods).

Here is a basic test program I am trying to run, it seems to be crashing on the `DERROR` call.

```c++
#include <iostream>
#include <memory>

#include <dji_vehicle.hpp>
#include <dji_linker.hpp>
#include <dji_log.hpp>

int main() {
  std::cout << "Hello world" << std::endl;
  // OsdkOsal_Free(nullptr);
  DJI::OSDK::Linker* l = new Linker();
  DJI::OSDK::Vehicle* v = new Vehicle(l);

  DJI::OSDK::Log::instance().enableStatusLogging();
  DJI::OSDK::Log::instance().enableDebugLogging();
  DJI::OSDK::Log::instance().enableErrorLogging();

  DERROR("Test error");

  std::cout << DJI::OSDK::Log::instance().getDebugLogState() << std::endl;
  std::cout << "End" << std::endl;

  std::cout << v->getFwVersion();
  // v->init();
  std::cout << "End" << std::endl;
  // std::unique_ptr<DJI::OSDK::Vehicle> v = std::make_unique<DJI::OSDK::Vehicle>()
}
```

And the stack trace from gdb if it helps:
```gdb
Program received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x000055a3ad221ec4 in OsdkOsal_MutexCreate ()
#2  0x000055a3ad1b04d2 in Mutex::Mutex (this=0x55a3ade37770)
    at /home/osdk-test/Onboard-SDK/osdk-core/platform/src/dji_platform.cpp:189
#3  0x000055a3ad1ac87f in DJI::OSDK::Log::title (this=0x55a3ade29eb0, level=1, 
    prefix=0x55a3ad22b051 "ERRORLOG", func=0x55a3ad22b04c "main", line=18)
    at /home/osdk-test/Onboard-SDK/osdk-core/logger/src/dji_log.cpp:61
#4  0x000055a3ad19e7fb in main () at /home/osdk-test/onboard/main.cpp:18
```

Are there some additional dependencies I need to get things working correctly?

dji-dev, 2020-09-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39188'>Zendesk ticket #39188</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Officially provided OSDK did not verify compatibility with Docker, we suggest that you use the official recommended Ubuntu1604 to use OSDK, according to the stack trace you provided, the error occurred in the linker lib of OSDK.

If you have requirements for docker, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzmhe, 2020-09-17:
We are running the OSDK ROS 4.0.1 and thus the OSDK 4.01 in a docker container.

arosendorff, 2020-09-17:
@lorenzmhe That's encouraging to hear!

Would you be willing to share your dockerfile for your container? And did you run into any issues with the linker code like I did?

mhoejgaard, 2020-09-17:
Unfortunately, I am not able to share the file at the moment as it's tightly coupled to our business, at least not without approval. I'll get back if I get the permission to do so. I'd like to share just a OSDK related excerpt but I cannot confirm if it actually works as our stack has a lot of other dependencies.

We did not encounter any issues specifically related to running in docker but it's not rare to see at least the legacy dji_sdk ROS node crash once in a while. This has drastically improved during the 3-4 years we have been using the OSDK ROS.

I can however tell you that we have simply installed the dependencies listed in the documentation and then
cloned the repo tag 4.0.1 and performed:
$ mkdir build
$ cd build
$ cmake .. -DADVANCED_SENSING=ON
$ make
$ make install

In 3.8.1 we did encounter an issue related to some lines in CMakeLists.txt that we had to remove to be able to build it.

closed 2021-06-24 by dji-kyle