dji-sdk/Onboard-SDK #634: Error in advanced sensing when other ACM devices present.

wiktorpi, 2020-09-16:
I have encountered what i believe to be a bug in dji_linker.
The system is TX2 running Ubuntu 18.04. Drone is Matrice M210 RTK.
We are using other sensors that appear as /dev/ttyACM0 and /dev/ttyACM1. 
The USB cable is NOT connected to the drone itself.
Using DJI-OSDK-ROS I launch dji_vehicle_node.launch and get the error:

> [2506641.564]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = **XXXXX**
[2506641.564]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[2506641.564]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2506641.565]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2506642.565]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[2506642.767]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2506642.771]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2506642.772]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2506642.773]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2506642.774]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1
[2506642.775]STATUS/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!
[2506642.783]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = **XXXXX**
[2506642.783]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[2506642.783]STATUS/1 @ AdvancedSensing, L145: Advanced Sensing init for the M210 drone
[2506642.784]STATUS/1 @ init, L49: Looking for USB device...
[2506642.800]STATUS/1 @ init, L65: Found 5 USB devices, identifying DJI device...
[2506642.801]ERRORLOG/1 @ init, L89: Did not find any DJI USB device, please double-check your connection.
[2506642.801]ERRORLOG/1 @ init, L91: Some helpful commands are 'lsusb' and 'dmesg'.
[2506642.802]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[2506643.302]STATUS/1 @ init, L254: Start advanced sensing initalization
[2506643.303]STATUS/1 @ activate, L1329: version 0x304032C
[2506643.303]STATUS/1 @ adv_pthread, L46: [2506643.303]STATUS/1 @ adv_pthread, L48: adv pthread running !!!!!!!!!!!!!!!!!!!!!!!
[vehicle_node-2] process has died [pid 18432, exit code -11, cmd /home/ubuntu/workspaces/dji_ws/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/home/ubuntu/.ros/log/4bf84fd6-f7de-11ea-aed9-00044bc77ff0/vehicle_node-2.log].
log file: /home/ubuntu/.ros/log/4bf84fd6-f7de-11ea-aed9-00044bc77ff0/vehicle_node-2*.log

Running with valgrind gives this:

>[2508355.688]STATUS/1 @ adv_pthread, L48: adv pthread running !!!!!!!!!!!!!!!!!!!!!!!
==23738== Thread 15:
==23738== Use of uninitialised value of size 8
==23738==    at 0x5178E78: ??? (in /lib/aarch64-linux-gnu/libusb-1.0.so.0.1.0)
==23738== 
==23738== Invalid read of size 8
==23738==    at 0x5178E78: ??? (in /lib/aarch64-linux-gnu/libusb-1.0.so.0.1.0)
==23738==  Address 0x48 is not stack'd, malloc'd or (recently) free'd
==23738== 
==23738== 
==23738== Process terminating with default action of signal 11 (SIGSEGV)
==23738==  Access not within mapped region at address 0x48
==23738==    at 0x5178E78: ??? (in /lib/aarch64-linux-gnu/libusb-1.0.so.0.1.0)
==23738==  If you believe this happened as a result of a stack
==23738==  overflow in your program's main thread (unlikely but
==23738==  possible), you can try to increase the size of the
==23738==  main thread stack using the --main-stacksize= flag.
==23738==  The main thread stack size used in this run was 8388608.

I have tracked the issue to be in osdk-core/api/src/dji_vehicle.cpp:246.
```
#ifdef ADVANCED_SENSING
  /*! If M300 here will use a new linker to do usb bulk
   * */
 ** if (!linker->isUSBPlugged()) { **
    DSTATUS( "USB is not plugged or initialized successfully. "
             "Advacned-Sensing will not run.");
  } else {
    if (!initAdvancedSensing()) {
      DERROR("Failed to initialize AdvancedSensing!\n");
      return false;
    } else {
      DSTATUS("Start advanced sensing initalization");
    }
  }
#endif
```
Seems that isUSBPlugged() still returns true even though the drone is not plugged in through USB.
I cannot debug further because there is only a header file available for dji_linker.

Best regards,
Wiktor

dji-dev, 2020-09-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39210'>Zendesk ticket #39210</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log shows that ：
[2506642.801]ERRORLOG/1 @ init, L89: Did not find any DJI USB device, please double-check your connection.

Have you added the DJI  USB Device node ? or it works normally when just one ACM device connects?
 [image]​

And suggest provide the readout with cmd "ifconfig".

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/tWKjd3wKZ4wLMhtu1rRH6vn9Y' target='zd_attach'>inline1605566240.png</a>

wiktorpi, 2020-09-17:
Yes, the DJI USB Device Node is already set up. The osdk works fine if the drone is connected to the USB or the other sensors appearing as /dev/ttyACMx are not plugged in.
As i said before, the issue is when the drone is NOT connected by USB cable to onboard computer, but other devices appearing as /dev/ttyACMx are. 
If i understand correctly, the line from log:
>[2506642.801]ERRORLOG/1 @ init, L89: Did not find any DJI USB device, please double-check your connection.

Means that the osdk correctly recognizes that no DJI device is present. However, the line:
>[2506643.302]STATUS/1 @ init, L254: Start advanced sensing initalization

Suggests that despite lack of connection, it is trying to start advanced sensing, which it should not do. That's what makes it crash i believe.

Best regards,
Wiktor

dji-dev, 2020-09-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39210'>Zendesk ticket #39210</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, as you said, is it possible to understand this? Your question is that the linker->isUSBPlugged() succeeded in connecting to other non-DJI ACM devices. The result you want is, isUSBPlugged should be set false.

If so , it will not affect actual use. About linker->isUSBPlugged() may be dependent on the ACM device name. It is recommended that you use a non-DJI ACM device for testing and verification by change the ACM device name, and the problem will be submitted to the relevant engineer to confirm whether it is an OSDK bug.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-kyle, 2020-10-19:
IsUSBPlugged () will return true even though the ACM device is not the DJI ACM device when the ACM name specified in userconfig.txt exists. This problem can be evaluated by setting the UserConfigtxt and ACM devices in the following manner：
Add： /etc/udev/rules.d/DJIDevice.rules ，with:
SUBSYSTEM=="usb", ATTRS

{idVendor}
=="2ca3", MODE="0666"
KERNEL=="ttyACM*", ENV

{ID_VENDOR_ID}
=="2ca3", ENV

{ID_MODEL_ID}
=="001f",MODE="0666", SYMLINK+="DJI_ACM"

and then, DJI's ACM device maps to /dev/dji_ACm, and UserConfig specifies the path

closed 2021-06-24 by dji-kyle