dji-sdk/Onboard-SDK #635: M300 can not get RGBImage from fpv camera on BR V4.0.1

dingpwen, 2020-09-16:
When trying to run camera-stream-poll-sample, I got:
 ./build/bin/camera-stream-poll-s
ample ./sample/platform/linux/common/UserConfig.txt
Please enter the type of camera stream you want to view(M210 V2)
m: Main Camera
f: FPV  Camera
b: both Cameras
f
argv[1] = ./sample/platform/linux/common/UserConfig.txt
Read App ID
User Configuration read successfully.

[2518233.833]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7800AS
J15
[2518233.833]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[2518233.833]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by
getting drone version.
[2518233.835]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2518234.835]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task
 created.
[2518235.036]STATUS/1 @ FileMgrImpl, L253: register download file callback handler su
ccessfully.
[2518235.037]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC
ground station status pushing : 1

[2518235.038]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[2518235.073]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7800AS
J15
le type:1
[2518236.904]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[2518236.906]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[2518236.907]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 16
9
[2518236.910]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[2518236.910]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[2518237.898]STATUS/1 @ activate, L1329: version 0x3040838

[2518237.932]STATUS/1 @ activate, L1367: Activation successful

STATUS/1All components for decoding initialized ...

[2518238.963]STATUS/1 @ startH264Stream, L464: Getting FPV Test
subsrcibe data success!
unsubsrcibe data success!
[2518259.079]STATUS/1 @ verify, L244: Verify subscription successful.
[2518259.099]STATUS/1 @ reset, L551: Reset Subscription Successful.

[2518259.403]STATUS/1 @ ~AdvancedSensing, L187: Destroy task for M300's liveview!
Segmentation fault (core dumped)

You will find that, it is success to connect to the fpv camera, but no RGBIMAGE got, if any, it will print:
void show_rgb(CameraRGBImage img, char* name)
{
  cout << "#### Got image from:\t" << string(name) << endl;

Please tell me that if any other settings are need.

dji-dev, 2020-09-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39217'>Zendesk ticket #39217</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Is this an exception using the original demo ?

I test the origin demo: camera-stream-poll-sple, with M300 RTK (firmware：v01.00.0211) + OSDK 4.0.1
it works normally.
 [image]​

Suggest try it again with original demo(don't change the demo code).

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/HN7K2Sj5XM4u6JVi3YCFnHqSR' target='zd_attach'>inline-474725506.png</a>

dingpwen, 2020-09-17:
I am sure that i am using the original demo, without any modification, The only matter is that my Ubuntu is 18.04.

dji-dev, 2020-09-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39217'>Zendesk ticket #39217</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log promts: Segmentation fault (core dumped), if it is convient , suggest you trace the stack error by GDB first.

and then the OSDK recommands to use Ubuntu1604. And you can check this issue with Ubuntu 1604.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dingpwen, 2020-09-17:
[2611455.545]STATUS/1 @ activate, L1367: Activation successful

[New Thread 0x7f9affce70 (LWP 29334)]
[New Thread 0x7f9a7fbe70 (LWP 29335)]
[New Thread 0x7f99ffae70 (LWP 29336)]
[New Thread 0x7f997f9e70 (LWP 29337)]
STATUS/1All components for decoding initialized ...

[2611456.567]STATUS/1 @ startH264Stream, L464: Getting FPV Test
subsrcibe data success!
[New Thread 0x7f98ff8e70 (LWP 29338)]
[Thread 0x7f9a7fbe70 (LWP 29335) exited]
[Thread 0x7f997f9e70 (LWP 29337) exited]
[Thread 0x7f99ffae70 (LWP 29336) exited]
[Thread 0x7f9affce70 (LWP 29334) exited]
unsubsrcibe data success!
[Thread 0x7f98ff8e70 (LWP 29338) exited]
[Thread 0x7fb1945e70 (LWP 29328) exited]
[2611476.704]STATUS/1 @ verify, L244: Verify subscription successful.
[2611476.724]STATUS/1 @ reset, L551: Reset Subscription Successful.

[Thread 0x7fb0943e70 (LWP 29330) exited]
[Thread 0x7f9bffee70 (LWP 29331) exited]
[Thread 0x7f937fde70 (LWP 29332) exited]
[2611477.031]STATUS/1 @ ~AdvancedSensing, L187: Destroy task for M300's liveview
!
[Thread 0x7fb1144e70 (LWP 29329) exited]

Thread 1 "camera-stream-p" received signal SIGSEGV, Segmentation fault.
0x00000055555c7700 in OsdkLinux_TaskDestroy (task=0x0)
    at /home/insky/project/FlyControl/Onboard-SDK/sample/platform/linux/osal/osd
kosal_linux.c:67
67        pthread_cancel(*(pthread_t *)task);
(gdb)

dingpwen, 2020-09-18:
I have got the reason! when I run as root, it is coming to ok now. I don't know why it should be running in root, when already added my user to dialout :
sudo usermod -a -G dialout $USER

dji-dev, 2020-09-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39217'>Zendesk ticket #39217</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

OK , the OSDK obtain stream via the ACM device, you can check the ACM device permission, "`ls -l/dev/xxx`",
and set with cmd : sudo chmod 777 /dev/xxx
​
Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2020-09-21 by dingpwen