dji-sdk/Onboard-SDK #638: "firewall logic" takes too long at initialization stage.

akamilkhan, 2020-09-20:
This piece of code takes around ~5 min at the start of applications, which is very slow in my opinions. If I comment this code out from the library, it seems like functionality is not affected.

**file path:** osdk-core/api/src/dji_vehical.cpp 
```
#if 1
  /*! M300 drone do the firewall logic */
  uint8_t retryTimes = 0;
  if (this->isM300() && this->linker->isUSBPlugged()) {
    firewall->setAppKey((uint8_t *) data->encKey, strlen(data->encKey) - 1);
    do {
      retryTimes ++;
      DSTATUS("osdk policy file updating(1) ......");
      OsdkOsal_TaskSleepMs(1000);
    } while ((!firewall->RequestUpdatePolicy()) && (retryTimes < 15));

    /*! pending for firewall logic finished */
    retryTimes = 0;
    do {
      retryTimes++;
      DSTATUS("osdk policy file updating(2) ......");
      OsdkOsal_TaskSleepMs(1000);
    } while ((!firewall->isPolicyUpdated()) && (retryTimes < 15));
  }
#endif
```

**Questions:**

1. Can someone explain what is the purpose of this code? 
2. why do we need a firewall in the first place? 
3. Is it safe to remove it to get faster initialization?

dji-dev, 2020-09-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39338'>Zendesk ticket #39338</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About this issue, I want to comfirn with you that if you connects the USB cable to the drone ?
or you can provide the terminal logs with takes around ~5 min to start the Application.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

akamilkhan, 2020-09-21:
Yes, when USB cable is connected only then it takes ~5 min at the start with "osdk policy file updating" logs.

dji-dev, 2020-09-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39338'>Zendesk ticket #39338</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, I'm still a little confused. 5 min looks like an exception, but it doesn't seem to be relevant to the code section you mentioned. The following test is for the relevant code execution time, which is about 2s.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/7iIzJJdSt9AQsNJV7v78MiL9l' target='zd_attach'>inline1593577341.png</a>

closed 2021-06-24 by dji-kyle