dji-sdk/Onboard-SDK #640: M300 OSDK - drone version not obtained

parazero, 2020-09-21:
Hi,
Since I cannot get the manifold 2 in my region, I created a cable according to https://forum.dji.com/forum.php?mod=viewthread&tid=219723.
The cable is connected via FTDI converter directly to the drone's OSDK port.
when connected to a serial terminal application I get data from the drone
[image]

When I try to debug the flightcontrol-sample in Linux, I get the following:
[image]

Data is also received in Linux using 'screen' command.
testing the cable for 'loopback' works fine.
I have changed the port baud rate to several options (230400 & 921600) to no avail.
the ACM cable is connected through an additional USB port to the drone's port directly.
Am I missing some HW components in my setup? 

I have entered all the relevant Linux commands to get the required permissions as advised in 
https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html
& 
https://developer.dji.com/onboard-sdk/documentation/quickstart/run-the-sample.html
Am I missing something in that department?

The final goal is to use STM32 as FC, but testing is easier using the Linux environment.
Any additional things I can test?
Are there other working setup designs I can try?
Thanks for your help.

akamilkhan, 2020-09-21:
Is this your first time using your drone with OSDK? You need to activate OSDK before you can run it.

dji-dev, 2020-09-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39366'>Zendesk ticket #39366</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

As the picture shows ,the picture shows that the serial communication is abnormal.
Please check the Serial cable connection and the baud rate(The OSDK config file: UserConfig.txt and the drone: DJI Assistant2 should keep the same baud rate).

And then maybe you should check the cable you made, The DJI adapter board has been released, if it is convient, suggest check it with DJI adapter board.
links, https://developer.dji.com/document/1bb1b2dc-9694-4d6b-ad43-fa69c5a10400
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/OMvprPKCckDB6w19rON8zbgYP' target='zd_attach'>inline-397834653.png</a>

Caufmann, 2020-10-19:
Hi Parazero,

I had the same problem as yours and fixed it in the UserConfig.txt .
Had to change Device from /ttyACM0 to /ttyTHS2 .

Since you didn't answer from a few days, hope you found a solution on your own.

See you,

parazero, 2020-11-02:
Hi Caufman,
Was not available for a few days
Thanks for your reply.
When I use ACM0 the printout shows that ACM0 link is created, I have no THS2 Device listed when using "dmseg | grep tty" command but have the ACM one.
can you please elaborate on your solution?
Thanks

closed 2021-06-24 by dji-kyle