dji-sdk/Onboard-SDK #642: Gimbal control

rebellojason7, 2020-09-29:
@dji-dev I am trying to control the gimbal using the GimbalAction service. When I publish a command of 0 degrees in the yaw axis, it goes to something completely arbitrary with regard to the drone heading. Can you tell me a way such that when I publish angles, its with regard to the drone heading ?

dji-dev, 2020-09-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39666'>Zendesk ticket #39666</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This may be related to the rotationMode you set,
0: absolute position control, and the coordinate is NED.
1:  relative to the current position for control.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

choiwooseok, 2020-10-15:
@rebellojason7, @dji-dev 

I think the prob is

setting.reference is hard coded with 0.... when call rotateAsync or rotateSync api

```cpp
// file : dji_gimbal_module.cpp
void GimbalModule::rotateAsync(Rotation rotation,
                               void (*userCB)(ErrorCode::ErrorCodeType retCode,
                                              UserData userData),
                               UserData userData) {
  if (getEnable()) {
    gimbalAngleSetting setting = {0};
    setting.yaw_angle = rotation.yaw * 10;
    setting.pitch_angle = rotation.pitch * 10;
    setting.roll_angle = rotation.roll * 10;
    setting.allowance = 10;  /*!< 0.1 degree allowance */
    setting.reference = 0; /*!< default reference (initial point) */
    setting.coordinate = rotation.rotationMode;
    setting.is_control = 1; /*!< take control */
    setting.timeout = 0; /*!< default 2s timeout */
    setting.time_for_action = rotation.time;
```

```
// file : dji_gimbal_module.hpp

gimbalAngleSetting

    /*! 0 = execute angle command based on the previously set reference point
     *  1 = execute angle command based on the current point
     */
    uint8_t reference               :1;
```

dji-kyle, 2021-07-07:
This issue has been fixed. Please update the master branch to verify this issue.

closed 2021-07-07 by dji-kyle