dji-sdk/Onboard-SDK #644: compile issue on Ubuntu 20.10

LorenzoMonti, 2020-10-11:
I tried to compile the Onboard-SDK on my Ubuntu 20.10 laptop but after launch ``make -j8``, it return this error:

[ 14%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/modules/src/filemgr/impl/mmap_file_buffer.cpp.o
In file included from /home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:4:
/home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:23:15: error: field ‘currentLogFilePath’ has incomplete type ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’}
   23 |   std::string currentLogFilePath;
      |               ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/10/iosfwd:39,
                 from /usr/include/c++/10/memory:74,
                 from /home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:13,
                 from /home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:4:
/usr/include/c++/10/bits/stringfwd.h:74:11: note: declaration of ‘std::string’ {aka ‘class std::__cxx11::basic_string<char>’}
   74 |     class basic_string;
      |           ^~~~~~~~~~~~
/home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:16:39: error: ‘path’ has incomplete type
   16 | bool MmapFileBuffer::init(std::string path, uint64_t fileSize) {
      |                           ~~~~~~~~~~~~^~~~
In file included from /usr/include/c++/10/iosfwd:39,
                 from /usr/include/c++/10/memory:74,
                 from /home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:13,
                 from /home/monti/Desktop/Onboard-SDK-3.3.1/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:4:
/usr/include/c++/10/bits/stringfwd.h:74:11: note: declaration of ‘std::string’ {aka ‘class std::__cxx11::basic_string<char>’}
   74 |     class basic_string;
      |           ^~~~~~~~~~~~
make[2]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/build.make:505: osdk-core/CMakeFiles/djiosdk-core.dir/modules/src/filemgr/impl/mmap_file_buffer.cpp.o] Error 1
make[2]: Leaving directory '/home/monti/Desktop/Onboard-SDK-3.3.1/build'
make[1]: *** [CMakeFiles/Makefile2:561: osdk-core/CMakeFiles/djiosdk-core.dir/all] Error 2
make[1]: Leaving directory '/home/monti/Desktop/Onboard-SDK-3.3.1/build'
make: *** [Makefile:130: all] Error 2

Any suggestions?

Thank you in advance

dji-dev, 2020-10-12:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/39902'>Zendesk ticket #39902</a>:<br><br>Hi. I am sorry to tell you that onboard 3.3.1 is too old to support. Please update your onboard SDK version to 3.9 or 4.0. In addition, we recommend running the onboard SDK on Ubuntu 16.04. There may be compatibility issues when running on Ubuntu 20.04

AchmadFathoni, 2021-05-06:
@LorenzoMonti  Have you tried to apply #740?

dji-kyle, 2021-06-25:
Update to latest firmware and OSDK version,  if the problem persists, please reopen this issue or post it on the following forum community.
https://djisdksupport.zendesk.com/hc/en-us/community/topics

closed 2021-06-25 by dji-kyle