dji-sdk/Onboard-SDK #645: Failed when test camera_stream with M300

dangviethieu, 2020-10-12:
Hello all, when tested camera_stream_pool_sample, I got the error:
Anyone got the same issue? Any help here would be greatly appreciated. Thanks in advance.
```
./camera-stream-callback-sample UserConfig.txt UserConfig.txt 
Please enter the type of camera stream you want to view (M210 V2)
m: Main Camera
f: FPV  Camera
f
argv[1] = UserConfig.txt
argv[2] = UserConfig.txt
Read App ID
User Configuration read successfully. 

[468146.126]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 
[468146.126]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[468146.126]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[468146.126]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[468147.126]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[468147.326]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[468147.326]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[468147.326]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[468147.366]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 
[468147.366]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[468147.368]ERRORLOG/1 @ AdvancedSensing, L108: Failed to initialize USB Bulk Linker channel for liveview!
[468147.368]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!
[468147.368]ERRORLOG/1 @ AdvancedSensing, L123: Failed to initialize USB Bulk Linker channel for perception!
[468147.368]ERRORLOG/1 @ AdvancedSensing, L130: Failed to create task for advanced sensing!
[468147.368]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[468147.368]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[468148.170]STATUS/1 @ init, L254: Start advanced sensing initalization
[468148.170]STATUS/1 @ activate, L1309: osdk policy file updating(1) ......
[468149.173]STATUS/1 @ activate, L1317: osdk policy file updating(2) ......
[468149.179]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[468149.179]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[468149.184]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[468149.184]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[468149.189]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[468149.189]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 169
[468149.194]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[468149.194]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[468150.174]STATUS/1 @ activate, L1329: version 0x3040838

[468150.203]STATUS/1 @ activate, L1353: Activation successful

STATUS/1All components for decoding initialized ...

STATUS/1User callback thread created successfully!

[468151.208]STATUS/1 @ startH264Stream, L464: Getting FPV Test
STATUS/1****** Decoder Callback Thread Start ******

subscribe data failed!
ackData:89
end
Failed to open selected camera
[468151.245]STATUS/1 @ verify, L244: [468151.263]STATUS/1 @ reset, L551: STATUS/1Decoder Callback Thread Stopped...

[468151.567]STATUS/1 @ ~AdvancedSensing, L186: Destroy task for M300's liveview!
Segmentation fault (core dumped)

```

dji-dev, 2020-10-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39928'>Zendesk ticket #39928</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log shows that the USB device open failed, It is recommanded that check the access permission, and test with root user

 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/DnZP5Wv11T4KbNzkaXYyo3buU' target='zd_attach'>inline479261213.png</a>

dangviethieu, 2020-10-21:
@dji-dev Thank you, I used root user and could get streaming from camera.

closed 2020-10-21 by dangviethieu

tompe17, 2022-04-06:
You can also add your user to the group disk.  I had to do that to get the ROS SDK to work.