dji-sdk/Onboard-SDK #652: 请问怎么获取飞机的方位角、俯仰角？

addriumruss, 2020-10-26:
飞机是M600, 
OSDK3.9, 
ubuntu系统

现在需要获取飞机的方位角、俯仰角。

谢谢！

dji-dev, 2020-10-26:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/40417'>Zendesk ticket #40417</a>:<br><br>您好，飞机的方位角俯仰角是通过四元数给出的。可以订阅 TOPIC_QUATERNION 四元数，然后将四元数转换成飞机的欧拉角。

addriumruss, 2020-10-26:
@dji-dev 感谢及时回应！ 
请问"可以订阅 TOPIC_QUATERNION 四元数，然后将四元数转换成飞机的欧拉角"这句话有没有对应的示例代码，谢谢!

lanyusea, 2020-10-26:
https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles#Euler_Angles_to_Quaternion_Conversion

可以参考这里

addriumruss, 2020-10-27:
多谢了！ 
似乎是国外网站，不太好打开。分享一下：

#define _USE_MATH_DEFINES
#include <cmath>

struct Quaternion {
    double w, x, y, z;
};

struct EulerAngles {
    double roll, pitch, yaw;
};

EulerAngles ToEulerAngles(Quaternion q) {
    EulerAngles angles;

    // roll (x-axis rotation)
    double sinr_cosp = 2 * (q.w * q.x + q.y * q.z);
    double cosr_cosp = 1 - 2 * (q.x * q.x + q.y * q.y);
    angles.roll = std::atan2(sinr_cosp, cosr_cosp);

    // pitch (y-axis rotation)
    double sinp = 2 * (q.w * q.y - q.z * q.x);
    if (std::abs(sinp) >= 1)
        angles.pitch = std::copysign(M_PI / 2, sinp); // use 90 degrees if out of range
    else
        angles.pitch = std::asin(sinp);

    // yaw (z-axis rotation)
    double siny_cosp = 2 * (q.w * q.z + q.x * q.y);
    double cosy_cosp = 1 - 2 * (q.y * q.y + q.z * q.z);
    angles.yaw = std::atan2(siny_cosp, cosy_cosp);

    return angles;
}

dji-dev, 2020-10-27:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/40417'>Zendesk ticket #40417</a>:<br><br>您好，我们提供的ROS代码中有四元数转方位角的示例。
 [image]​<br><a href='https://djisdksupport.zendesk.com/attachments/token/uJxm2Tp5Ee0W1qXtCTeN4JUYz' target='zd_attach'>inline-1276900615.png</a>

addriumruss, 2020-10-28:
@dji-dev 你好，上面提供的这个图片完全看不懂，从quat怎么转成yaw,pitch,roll?
另外按照https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles#Euler_Angles_to_Quaternion_Conversion执行的计算后，获取到方位总是3.14, 移动飞机位置、方向也没有变化，请问是否还有什么注意事项？这个结果是弧度还是角度呢？ 
谢谢！

dji-dev, 2020-10-30:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/40417'>Zendesk ticket #40417</a>:<br><br>您好，这是调用了ROS里面的tf包进行的转换，您可以查阅相关的资料，也可以到ROS包里面阅读完整的代码。

closed 2021-06-25 by dji-kyle