dji-sdk/Onboard-SDK #653: 关于航点之间转向问题

addriumruss, 2020-10-27:
环境: M600, OSDK3.9, Ubuntu
现实场景： 
通过osdk设置航线，包含多个航点，由于航点自身没有悬停属性, 现在通过侦听航点到达事件在waypoint_reached_callback_handler回调里启动子线程，在子线程里执行wpMission->pause悬停飞机。悬停之后做拍照、录像等各种处理。处理结束后在调用wpMission->resume恢复航线飞行。
问题点：
飞机在悬停之前自动改变了方向，在通过上述方式停住飞机之前，飞机头部已经转到下一个航点方向，导致拍照、录像等操作的角度不对。

请问有没有办法在飞机头部方位改变之前准确停住飞机？ 在wpMission->resume之后恢复飞行了飞机再自动改变方位飞往下一个航点？

谢谢！

dji-dev, 2020-10-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40473'>Zendesk ticket #40473</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

航点任务中可以考虑通过参数设置航点执行任务功能，如下：
 [image]​

对应设置action的参数说明：
 [image]​
感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/dqYFDRnGPht7lgraymhX2B4DB' target='zd_attach'>inline-1577366364.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/NGXhWkhOSAsmmTbaTaPMAMFdB' target='zd_attach'>inline1311826835.png</a>

addriumruss, 2020-10-29:
@dji-dev 你好，感谢你的回复！
但是以上设置中"拍照"已标明"动作参数无作用"，而且就算它有作用也不管用，我这边是第三方镜头，需要从机载嵌入式板中对镜头rtsp流进行拍照然后自动上传到云端，不是存储飞机SD卡！
我这边现实场景重点在于悬停，希望悬停之前飞机不要改变头部方位，请问是否有办法解决这个问题？

dji-dev, 2020-10-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40473'>Zendesk ticket #40473</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

您所提到的问题应该是在到达航点回调时候，机头已经转向。机头YAW值可以通过相关参数设置：
typedef struct WayPointInitSettings
{
...
  uint8_t yawMode; /*!< Yaw mode <br>*/
  /*!< 0: auto mode(point to next waypoint) <br>*/
  /*!< 1: lock as an initial value <br>*/
  /*!< 2: controlled by RC <br>*/
  /*!< 3: use waypoint's yaw(tgt_yaw) */
...
} WayPointInitSettings; // pack(1)

typedef struct WayPointSettings
{
...
  int16_t  yaw;    /*!< Yaw (degree) */
...
} WayPointSettings;              // pack(1)

该部分可以根据您的实际需求来设定。

另外，根据您的描述，也可以评估通过应用端来实时获取定位位置，通过应用端主动判断航点是否已经达到来调整控制。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

addriumruss, 2020-10-30:
好的 我从应用端想想办法 感谢支持！

closed 2021-06-25 by dji-kyle