dji-sdk/Onboard-SDK #654: Does SDK has bare-metal version

Mahdi-Hosseinali, 2020-10-27:
I'm looking at using OSDK for a project and was wondering if I want to use stm32 do I have to use RTOS or the bare-metal version of the OSDK also exists? Obviously, I don't want to port the OSDK to bare-metal.

dji-dev, 2020-10-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40494'>Zendesk ticket #40494</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The OSDK version 4.0 for STM32 the RTOS should be used . The OSDK 3.9 can used for the bare-metal.
For OSDK version compatible drones, please refer to the developer's official website
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/bS3CdIJygh6hvoImfKsMqMVQa' target='zd_attach'>inline-1472604441.png</a>

Mahdi-Hosseinali, 2020-10-28:
But they are totally different drone support. So if I want to use the OSDK with say M210 V2 then I can only use version 4.0 which is RTOS, right?

dji-dev, 2020-10-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40494'>Zendesk ticket #40494</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Yes， OSDK4.0 needs to be used with RTOS/Linux/ROS.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-25 by dji-kyle