dji-sdk/Onboard-SDK #657: WayPointV2中（M300）到达航点如何悬停？

tuyaliang, 2020-11-04:
WayPointV2中（M300）到达航点如何悬停？之前waypoint有stay操作

tuyaliang, 2020-11-05:
`std::vector<DJIWaypointV2Action> WaypointV2MissionSample::generateWaypointActions(uint16_t actionNum)
{
  std::vector<DJIWaypointV2Action> actionVector;

  for(uint16_t i = 0; i < actionNum; i++)
  {
    DJIWaypointV2SampleReachPointTriggerParam sampleReachPointTriggerParam;
    sampleReachPointTriggerParam.waypointIndex = i;
    sampleReachPointTriggerParam.terminateNum = 0;
    
    auto *FlyingParam = new DJIWaypointV2AircraftControlFlyingParam;
    FlyingParam->isStartFlying = false;
    
    auto *trigger = new DJIWaypointV2Trigger(DJIWaypointV2ActionTriggerTypeReachPoint,&sampleReachPointTriggerParam);
    auto *FlyingActuatorParam = new DJIWaypointV2AircraftControlParam(DJIWaypointV2ActionActuatorAircraftControlOperationTypeFlyingControl, FlyingParam);
    auto *actuator = new DJIWaypointV2Actuator(DJIWaypointV2ActionActuatorTypeAircraftControl, 0, FlyingActuatorParam);
    auto *action = new DJIWaypointV2Action(i, *trigger,*actuator);
    actionVector.push_back(*action);

  return actionVector;
}`

尝试加入悬停的代码，没有生效

dji-dev, 2020-11-05:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/40743'>Zendesk ticket #40743</a>:<br><br>您好
建议下载最新的SDK试试，之前有修复过不能悬停的问题。此外悬停是停止和恢复两个动作联合使用的，中间需要加一定时间的延时，请检查下您的代码是不是这样编写的。
​

tuyaliang, 2020-11-06:
> Agent comment from Hummels Lei in [Zendesk ticket #40743](https://djisdksupport.zendesk.com/agent/tickets/40743):
> 
> 您好
> 建议下载最新的SDK试试，之前有修复过不能悬停的问题。此外悬停是停止和恢复两个动作联合使用的，中间需要加一定时间的延时，请检查下您的代码是不是这样编写的。
> ​

这是错误log，帮看一下什么原因
[2620567.398]STATUS/1 @ uploadWaypointMission, L318: Upload waypoint v2 mission successfully!
[2620568.402]STATUS/1 @ downloadWaypointMission, L334: Download waypoint v2 mission successfully!
[AIOST] ========== action takephoto!
[2620569.406]ERRORLOG/1 @ uploadWapointActions, L344: Upload waypoint v2 actions ErrorCode:0x110000420003
[2620569.406]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : WaypointV2
[2620569.406]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : the Function ID of Actuator is invalid. It might not defined or the information is empty. 
[2620569.406]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : none
[2620570.413]STATUS/1 @ updateMissionEvent, L83: MissionEventPushAck->event ID :0x0
[2620570.413]STATUS/1 @ startWaypointMission, L365: Start waypoint v2 mission successfully!

dji-kyle, 2021-06-25:
参考社区帖：
https://djisdksupport.zendesk.com/hc/zh-cn/articles/4402367493785-OSDK-waypoint-v2-sample%E4%BB%A3%E7%A0%81%E8%B7%9F%E8%AF%BB

closed 2021-06-25 by dji-kyle