dji-sdk/Onboard-SDK #658: Error occur when reload missions have actions

dangviethieu, 2020-11-09:
Hi all, I created one plan from waypoint_v2_sample.cpp file, I got this issue when try reload (or upload new mission).
[image]
Any suggestions?
Thank you in advance

dji-dev, 2020-11-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40861'>Zendesk ticket #40861</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This error looks like you upload actions parameter error, it is recommended that you check the relevant parameters,
if it still cannot be resolved, you can provide the relevant code.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-11-10:
@dji-dev Thank you for reply. But my problem is in the first upload mission, system can load and start mission successfully, then stop mission, I want reload that mission again, (or continue upload new mission), don't restart dji_osdk_ros node, the above issue will occur.
Are you test with this case?

dji-dev, 2020-11-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40861'>Zendesk ticket #40861</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

When uploading again after the stop task, you need to do “init ” first，
Do you have an init mission when you upload again?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-11-11:
@dji-dev Yes, I did "init" mission successfully!
Like the image I uploaded: https://user-images.githubusercontent.com/14278358/98495962-f8c1d200-2272-11eb-8604-bc0fe83bdc71.png
the system did upload actions, init mission, upload mission, download mission successfully, but failed at upload actions.

dji-dev, 2020-11-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40861'>Zendesk ticket #40861</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I test to reload action success ,code:
 [image]​
and result:
 [image]​

please check the code again , If the problem persists, suggest you provide the code for confirmation.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/fcJg243YTZEblBW88N8JYo4Gv' target='zd_attach'>inline623883654.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/yIWrYAMB1Lw0axoitx7fkjRIO' target='zd_attach'>inline-187251970.png</a>

dangviethieu, 2020-11-13:
@dji-dev Thank you for confirmed.
The issue occurred when I use ROS, I investigated in source code, at `dji_vehicle_node_mission_services.cpp` file, when generate new actions, they pushed back the new actions, but don't clear it before.
The solution is clear `this->actions` before generate new actions, or clear it after upload action successfully.
[image]

closed 2020-11-13 by dangviethieu