dji-sdk/Onboard-SDK #660: mop api questions

choiwooseok, 2020-11-10:
Hi.

I have a few questions in Onboard-SDK 4.0.1 mop apis.

    Is there any plan with non-blocking api support?

    when recv or send return code -> MOP_CONNECTIONCLOSE
    -> Then I want to close and reconnect with Mobile App.
    -> So I call MopSever::close with pipeline ID.
    -> return code is MOP_FAILED.. but why? need help..
    -> I think in pipelineMap clear logics are not correctly called.

    when connection closed by App close or move to connection lost area -> I got return code 'MOP_UNKNOWN_ERR '
    -> In this time same as above I want to close and reconnect with Mobile App.

Please give me help! Thanks.

dji-dev, 2020-11-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40896'>Zendesk ticket #40896</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Is there any plan with non-blocking api support?
--> Sorry,there is currently no such information, you can continue to follow the OSDK related function updates.

when recv or send return code -> MOP_CONNECTIONCLOSE
-> Then I want to close and reconnect with Mobile App.
-> So I call MopSever::close with pipeline ID.
-> return code is MOP_FAILED.. but why? need help..
-> I think in pipelineMap clear logics are not correctly called.
when connection closed by App close or move to connection lost area -> I got return code 'MOP_UNKNOWN_ERR '
-> In this time same as above I want to close and reconnect with Mobile App.

==》This issue we will verify it , and if there is any result , we will update this issue.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-12-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40896'>Zendesk ticket #40896</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About the issue: mop api questions
The problem has been fixed and it has been updated to the master branch. You can update and verify it.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

choiwooseok, 2020-12-23:
Hi @dji-dev 

I have been test with master branch. It's not working ...

Test scenario :
    disconnect by app
    and then
    Onboard : call accecpt with pipeline id
    App : call connect

I got return code 'MOP_RESOCCUPIED'

in here

```cpp
  /*! 2.Do binding */
  DSTATUS("/*! 2.Do binding */");
  ret = mop_bind_channel(bind_handle, id);
  if (ret != MOP_SUCCESS) {
    DERROR("MOP Pipeline bind failed");
    return getMopErrCode(ret);
  }
  ```

dji-kyle, 2021-06-25:
Please update to OSDK 4.1, the sample code has been updated.

closed 2021-06-25 by dji-kyle

hieudv-terradrone, 2021-11-11:
`About the issue: mop api questions`
when are you support it?