dji-sdk/Onboard-SDK #669: OSDK 4.0.1 FPVCameraStream Issue

choiwooseok, 2020-11-18:
vehicle : M300 RTK
OSDK : 4.0.1

```
  static char camName[] = "FPV_CAM";
  vehicle_->advancedSensing->startFPVCameraStream(&VisionImageCallback, &camName);
```

After User Callback Thread created

our Callback does not called..

I follow the osdk source code.

And I found my program stuck with 

```
void DJICameraStreamDecoder::callbackThreadFunc()
{
  while(cbThreadIsRunning)
  {
    CameraRGBImage copyOfImage;
    if(!decodedImageHandler.getNewImageWithLock(copyOfImage, 1000))
    {
      DDEBUG_PRIVATE("Decoder Callback Thread: Get image time out\n"); // <- This Line
      continue;
    }

    if(cb)
    {
      (*cb)(copyOfImage, cbUserParam);
    }
  }
  DSTATUS_PRIVATE("Decoder Callback Thread Stopped...\n");
}

```

But the camera-stream-callback-sample works...

Where sholud I have to check?

dji-dev, 2020-11-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41183'>Zendesk ticket #41183</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I want to confirm with you that using origin sample is normal. Is it your own application code that is abnormal?
If so，Please check the Setup steps:
 // Setup OSDK.
  bool enableAdvancedSensing = true;
  LinuxSetup linuxEnvironment(argc, argv, enableAdvancedSensing);
  Vehicle*  vehicle = linuxEnvironment.getVehicle();
  const char *acm_dev = linuxEnvironment.getEnvironment()->getDeviceAcm().c_str();
  vehicle->advancedSensing->setAcmDevicePath(acm_dev);

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

choiwooseok, 2020-11-19:
Yes I called a same 'Setup' routine.

but 'while(cbThreadIsRunning)' while loop never called cb. It keeps Timeout continues to occur.

choiwooseok, 2020-11-19:
```
    std::string configPath = PathUtil::GetHome().string() + "/UserConfig.txt";
    char* argv[3] = {(char*)"", (char*)(configPath.c_str()), (char*)"/dev/ttyACM0"};
    setup_ = std::make_unique<LinuxSetup>(3, argv, true);
    vehicle_ = setup_->getVehicle();
    const char* acm_dev = setup_->getEnvironment()->getDeviceAcm().c_str();
    vehicle_->advancedSensing->setAcmDevicePath(acm_dev);
```

I think setup routine is same.

but still infinite timeout occurs in Decoder callback thread

dji-dev, 2020-11-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41183'>Zendesk ticket #41183</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, the problem cannot be confirmed for the time being. If there is no problem with the sample, only the application code is abnormal.
It is recommended that you provide a complete terminal running log or complete modified code to view.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

choiwooseok, 2020-11-20:
```
argv[1] = /home/dji/UserConfig.txt
Find the target ttyACM device , path:/dev/ttyACM0
Read App ID
User Configuration read successfully.
Set ACM device:/dev/ttyACM0
[3815918.851]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7T00A69VH
[3815918.851]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3815918.851]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3815918.851]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2020-11-20 11:18:40.783] [info] Start deserializing ...
[3815919.851]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3815920.051]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3815920.052]STATUS/1 @ m300LensCbInit, L99: ...... register result of geting camera pushing : 1
[3815920.059]STATUS/1 @ m300LensCbInit, L118: Request start pushing lens info ack = 0
[3815920.059]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1
[3815920.060]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[3815920.060]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[3815921.067]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[3815921.076]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[3815921.076]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[3815921.077]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3815921.077]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[3815921.080]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3815921.080]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 169
[3815921.081]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[3815921.081]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[3815922.067]STATUS/1 @ firewallTask, L112: firewall task created ...
[3815922.082]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7T00A69VH
[3815922.082]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3815922.092]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[3815922.093]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[3815922.093]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[3815922.099]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[3815922.099]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3815922.099]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3815922.899]STATUS/1 @ init, L254: Start advanced sensing initalization
[3815922.899]STATUS/1 @ activate, L1329: version 0x3040838
[3815922.921]STATUS/1 @ activate, L1367: Activation successful
[3815923.942]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7T00A69VH
[3815923.942]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3815923.942]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3815923.942]STATUS/1 @ init, L254: Start advanced sensing initalization
[3815923.942]STATUS/1 @ activate, L1329: version 0x3040838
[3815923.962]STATUS/1 @ activate, L1367: Activation successful
[3815925.968]STATUS/1 @ accept, L48: /*! 0.Find whether the pipeline object is existed or not */
[3815925.968]STATUS/1 @ accept, L54: /*! 1.Create handler for binding */
[3815925.968]STATUS/1 @ accept, L62: /*! 2.Do binding */
[3815925.968]STATUS/1 @ accept, L75: /*! 3.Do accepting */
[3815925.968]STATUS/1 @ accept, L76: Do accepting blocking for channel [49154] ...
[3815928.939]STATUS/1 @ accept, L84: /*! 4.Accept finished */
[3815928.939]STATUS/1 @ accept, L86: MOP channel [49154] accepted success
[2020-11-20 11:18:50.347] [info] [FcIODjiImpl] MOP SEND Pipeline accept success
[3815928.939]STATUS/1 @ accept, L48: /*! 0.Find whether the pipeline object is existed or not */
[3815928.939]STATUS/1 @ accept, L54: /*! 1.Create handler for binding */
[3815928.940]STATUS/1 @ accept, L62: /*! 2.Do binding */
[3815928.940]STATUS/1 @ accept, L75: /*! 3.Do accepting */
[3815928.940]STATUS/1 @ accept, L76: Do accepting blocking for channel [49155] ...
[3815929.073]STATUS/1 @ accept, L84: /*! 4.Accept finished */
[3815929.073]STATUS/1 @ accept, L86: MOP channel [49155] accepted success
[2020-11-20 11:18:50.481] [info] [FcIODjiImpl] MOP RECV Pipeline accept success
[3815929.082]STATUS/1 @ verify, L244: Verify subscription successful.
[3815929.114]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3815929.114]STATUS/1 @ startPackage, L347: Package 1 info: freq=1, nTopics=4.
[3815929.122]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[3815929.122]STATUS/1 @ startPackage, L347: Package 2 info: freq=5, nTopics=3.
[3815929.142]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[3815929.142]STATUS/1 @ startPackage, L347: Package 3 info: freq=50, nTopics=2.
[3815929.163]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[3815929.163]STATUS/1 @ startPackage, L347: Package 4 info: freq=100, nTopics=4.
[2020-11-20 11:18:50.594] [info] [FcIODjiImpl] Succeed to release control auth
[3815929.186]STATUS/1 @ config, L96: sent
[... cut ...]
STATUS/1All components for decoding initialized ...
STATUS/1User callback thread created successfully!
[3815929.225]STATUS/1 @ startH264Stream, L464: Getting FPV Test
subsrcibe data success!
[2020-11-20 11:18:50.636] [info] [DJICamera] init camera module main camera success
[2020-11-20 11:18:50.636] [info] [DJICamera] init camera module vice camera success
[2020-11-20 11:18:50.636] [info] [DJICamera] init camera module top camera success
STATUS/1****** Decoder Callback Thread Start ******
```

and timeouts ...

dji-dev, 2020-11-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41183'>Zendesk ticket #41183</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

There are some differences between M210 and M300 to get the video stream, it is recommended to confirm whether the calling API is wrong
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/MBv1hFSI4T6wsCgwAXXDB3OnI' target='zd_attach'>inline-1251108705.png</a>

choiwooseok, 2020-11-20:
yes I have been saw that lines. I know in osdks implements are different.

And In my situation startFPVCameraStream returns 'true'

> There are some differences between M210 and M300 to get the video stream, it is recommended to confirm whether the calling API is wrong

The api that the caller calls is the same.

```
// SETUP
  vehicle_ = setup_->getVehicle();
  const char* acm_dev = setup_->getEnvironment()->getDeviceAcm().c_str();
  vehicle_->advancedSensing->setAcmDevicePath(acm_dev);

// START FPV Camera Stream
  vehicle_->advancedSensing->startFPVCameraStream(&VisionImageCallback, &camName);

// EXPECT CALLBACK Called
```

but in m300, always timeout occured... never called (*cb)(copyOfImage, cbUserParam)
```
void DJICameraStreamDecoder::callbackThreadFunc()
{
  while(cbThreadIsRunning)
  {
    CameraRGBImage copyOfImage;
    if(!decodedImageHandler.getNewImageWithLock(copyOfImage, 1000))
    {
      DDEBUG_PRIVATE("Decoder Callback Thread: Get image time out\n"); // <- This Line
      continue;
    }

    if(cb)
    {
      (*cb)(copyOfImage, cbUserParam);
    }
  }
  DSTATUS_PRIVATE("Decoder Callback Thread Stopped...\n");
}
```

dji-dev, 2020-11-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41183'>Zendesk ticket #41183</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log shows that the MOP function is being used at the same time, is it possible to have something to do with the use of this part?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

choiwooseok, 2020-11-20:
> The log shows that the MOP function is being used at the same time, is it possible to have something to do with the use of this part?

Yes. Exclude when connection lost situation. I use MOP api with read and write thread for telemetry & command.

dji-dev, 2020-11-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41183'>Zendesk ticket #41183</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Yes, it is recommended to investigate the impact of the use of the application function.if it still does not work, if it is convenient, you can provide the code to confirm the problem.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-25 by dji-kyle