dji-sdk/Onboard-SDK #676: keil v5 could not build osdk-stm32-sample. How to build it on keil?

xuehuxs, 2020-11-24:
keil v5 could not build osdk-stm32-sample. How to build it on keil?

errors:
[image]

dji-dev, 2020-11-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41406'>Zendesk ticket #41406</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

please check the ARMCC version and STM32F4xx_DFP.2.xx Driver package
1、Keil MDK 5.22 and above (armcc 5.06)
2、STM32F4 Pack version needs to use 2.13.0

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

xuehuxs, 2020-11-25:
ok,thanks!

closed 2021-06-25 by dji-kyle