dji-sdk/Onboard-SDK #677: H264 video does not have PTS information. How to get the pts information?

lonl, 2020-11-25:
Hi,
    Through the osdk demo, we save the H264 video and decode it, but could not get the pts information, how could I get the pts information?

dji-dev, 2020-11-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41442'>Zendesk ticket #41442</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Could you please provide the Camera type you obtained the video stream via OSDK?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lonl, 2020-11-25:
@dji-dev 
  Hi, the main camera we use is zenmuse H20T.
  The main camera video and FPV video both do not have pts info via osdk.
  Thank you for your replay!

dji-dev, 2020-11-26:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41442'>Zendesk ticket #41442</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

You can use ffmpeg to convert it to mp4 file:
ffmpeg.exe -flags2 showall -i input.h264 output.mp4

and then :
ffprobe.exe -show_packets output.mp4

you will get the PTS.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lonl, 2020-11-26:
Dear dji-dev, 

    Thank you very much ~

Best Regards,

closed 2020-11-26 by lonl