dji-sdk/Onboard-SDK #698: Zoom control of the camera mounted on the matrice600pro using OSDK 3.9

yoshida072, 2020-12-26:
The following sample code does not work for zoom in and zoom out. https://github.com/dji-sdk/Onboard-SDK/blob/3.9/sample/platform/linux/camera-gimbal/camera_ gimbal_sample.cpp#L303

Gimbal control using the following sample code is successful. https://github.com/dji-sdk/Onboard-SDK/blob/3.9/sample/platform/linux/camera-gimbal/camera_gimbal_sample.cpp#L39

Does anyone know how to do zoom in and zoom out with OSDK 3.9? 
Thanks for your help.

dji-dev, 2020-12-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42512'>Zendesk ticket #42512</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Regarding the code you provided, L303 and L39, which correspond to different functions.

please refer to the main function notes.
https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/sample/platform/linux/camera-gimbal/main.cpp#L55
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/Y93wb4d8uH59UT0bnFdF8a1eE' target='zd_attach'>inline1167058519.png</a>

yoshida072, 2020-12-28:
Thank you for your answer.
I could not zoom by typing a or b on the keyboard.
Do the M600PRO and Z30 support zooming via OSDK?

Agent comment from kyle.cai in Zendesk ticket #42512<https://djisdksupport.zendesk.com/agent/tickets/42512>:

https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/sample/platform/linux/camera-gimbal/main.cpp#L55
[https://camo.githubusercontent.com/0240492623e78ccd94183e8baa5417c3a5cc6dbedf75ecf41dc2b391fa79ddfb/68747470733a2f2f646a6973646b737570706f72742e7a656e6465736b2e636f6d2f6174746163686d656e74732f746f6b656e2f593933776234643875483539555430626e46644638613165452f3f6e616d653d696e6c696e65313136373035383531392e706e67]<https://camo.githubusercontent.com/0240492623e78ccd94183e8baa5417c3a5cc6dbedf75ecf41dc2b391fa79ddfb/68747470733a2f2f646a6973646b737570706f72742e7a656e6465736b2e636f6d2f6174746163686d656e74732f746f6b656e2f593933776234643875483539555430626e46644638613165452f3f6e616d653d696e6c696e65313136373035383531392e706e67>​

inline1167058519.png<https://djisdksupport.zendesk.com/attachments/token/Y93wb4d8uH59UT0bnFdF8a1eE>

closed 2021-06-25 by dji-kyle