dji-sdk/Onboard-SDK #703: What version of FFMPEG does OSDK4.0.1 rely on

github4529, 2021-01-08:
I Use the newest version FFMPEG cannot support. Maybe it depends on a lower version of FFMPEG?

github4529, 2021-01-08:
MANIFOLD 2G with FFMPEG without NVENC-support.
I'd like compile FFMPEG with code.But the version ...

dji-dev, 2021-01-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42858'>Zendesk ticket #42858</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The OSDK demo rely on ffmpeg version 2.8.15，If you use the newest version ffmpeg, you may need to modify the implementation of demo related decoding functions.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/wyQdDKRDEODt7AaqD6JsgSEbW' target='zd_attach'>inline1411065446.png</a>

github4529, 2021-01-14:
nvidia-smi not found 
DJI JETSON TX2 without nvidia driver?
I see it with CUDA9.0

dji-kyle, 2021-06-25:
Please consult the support@dji.com for more about manifold 2 production info.

closed 2021-06-25 by dji-kyle