dji-sdk/Onboard-SDK #704: Listing files on M300/H20T with OSDK 4.0.1

lorenz-ckp, 2021-01-08:
Hi,

I've been successful in using the functionality for listing files on the M300/H20T using the OSDK 4.0.1.
But when the amount of images on the SD card exceeds 11, It seems like the OSDK is not able to list the rest. Note that this is done on a freshly formatted SDK card where nothing else than single wide/zoom images were taken.

I did a small test to showcase it, and the result can be seen on the image below.

1. Take 15 images and list them using the OSDK.
2. Notice how there only seems to be listed 11 images in the logs.
3. Delete the first 8 images and list them again.
4. Notice how the last 3 images (13 - 15) are now shown together with the images from last listing giving a total of 7 images.
5. Take 10 more images and list again.
6. Notice how the count in the filenames only goes to 19, giving a total of 11 images, even though there should be 25 - 8 = 17.
7. Delete 7 files with filename count of 14 to 21 and list.
8. Notice how all images on the SD card are visible, giving a total of 9 images.
9. Take 5 more images, resulting in the last image having the filename count 30, and list the files.
10. Notice how the last three images are not shown, since the would exceed the limit of 11 images.

[image]

dji-dev, 2021-01-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42860'>Zendesk ticket #42860</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

We have noticed that there are problems with download files in version 4.0.1, and optimizations have been made on the master branch. You can download the master branch first to see whether the problem still exists
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/hUs4FRN9OQx3aX1ottu3oEhDZ' target='zd_attach'>inline-62995442.png</a>

lorenzmhe, 2021-04-28:
I can confirm that this was fixed in 4.1.0. 

However, everything fails if a highres grid is present on the SD-card: #764.

closed 2021-06-25 by dji-kyle