dji-sdk/Onboard-SDK #710: M300 with Manifold 2-C and OSDK Expansion module - Failed to initialize Linker channel

tobiasfriden, 2021-01-14:
Hello,

I want to develop an onboard SDK solution for DJI M300. I have the Manifold 2-C onboard computer together with the expansion module for easy installation.

I have connected the cable to the OSDK port on the drone and then the USB cable from expansion module to one of the USB-ports to the Manifold. I have also done the following steps from the tutorial:
`sudo usermod -a -G dialout $USER`
Create the `DJIDevice.rules` in `/etc/udev/rules.d`
Enabled OSDK in Assistant as shown in the picture:
[image]

I can see that there is a device created called `/dev/ttyACM0` and `dmesg` output:
[image]
However there are no devices called `/dev/ttyUSB*`

I use the following UserConfig.txt:
```
app_id : <my app id>
app_key : <my app key>
baudrate : 921600
acm_port : /dev/ttyACM0
device : /dev/ttyUSB0
```

And the example gives the following output:
```
dji@manifold2:~/Onboard-SDK/build/bin$ ./djiosdk-telemetry-sample UserConfig.txt UserConfig.txt 
argv[1] = UserConfig.txt
argv[2] = UserConfig.txt
Read App ID
User Configuration read successfully. 

[2945.211]ERRORLOG/1 @ initVehicle, L215: Failed to initialize Linker channel
Vehicle not initialized, exiting.
```

What am I missing? Any help much appreciated.

tobiasfriden, 2021-01-14:
If I set `device` to `/dev/ttyACM0` instead I get output:
```
[4255.318]STATUS/1 @ getDroneVersion, L1702: ret = 2
[4255.318]ERRORLOG/1 @ getDroneVersion, L1707: Drone version not obtained! Please do not proceed.
Possible reasons:
        Serial port connection:
                * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
                * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
                * TX and RX pins are invert
```

dji-dev, 2021-01-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43040'>Zendesk ticket #43040</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

 The log shows that the serial port initial failed, please check the hardware connections. if it is convient, you can provide the pictures about the connections and the readout with "ls -l | grep ttyUSB*"
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/1o70WDtlOLtaBt0StV6r3tSa4' target='zd_attach'>inline-593124260.png</a>

tobiasfriden, 2021-01-15:
Hello,

Connection of USB port of the extension module:
[image]

UART connection to the OSDK port:
[image]

Terminal output:
```bash
dji@manifold2:~$ ls -l /dev/ttyACM*
crw-rw---- 1 root dialout 166, 0 Jan 15 12:09 /dev/ttyACM0
dji@manifold2:~$ ls -l /dev/ttyUSB*
ls: cannot access '/dev/ttyUSB*': No such file or directory
```

So, indeed there is no /tty/USB* available (as I also wrote in my last post). I assume then that it is necessary to connect also the UART port and not only USB of the extension module? This is quite unclear in the documentation.

dji-dev, 2021-01-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43040'>Zendesk ticket #43040</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、For manifold 2C, please using USB-TTL cable to connects the manifold with drone. The UART0 used for debug. if the USB(-ttl) connects the 2C, it will be recongnized as ttyUSB*
 [image]​

2、The ACM device corresponds to the double A USB cable and has been recognized.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/VXdwheBQj5aIObM2NxOLNbP6U' target='zd_attach'>inline-788354969.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/hbWA7NBfm8yMgENyq3vk9dUK7' target='zd_attach'>inline-855481404.png</a>

tobiasfriden, 2021-01-15:
Hello, I see.
I think the documentation could definitely be clarified on this point (it is not clearly mentioned at all that the UART connection is necessary for M300).
Also, why is no such cable included in the Manifold 2-C kit if it is necessary for connection with the drone?

tobiasfriden, 2021-01-28:
Working with USB-TTL cable

closed 2021-01-28 by tobiasfriden

BenKatanHighLander, 2021-03-03:
@tobiasfriden please share your final setup

tobiasfriden, 2021-03-03:
> @tobiasfriden please share your final setup

@BenKatanHighLander USB-TTL cable connected to the serial OSDK port on the extension board, and another USB connected to the usb port on the board. Both connected to usb ports on the manifold

BenKatanHighLander, 2021-03-04:
@tobiasfriden i will appreciate it if you share some phots

abisaz, 2022-10-11:
@tobiasfriden, I would also appreciate a photo of your setup