dji-sdk/Onboard-SDK #712: How to capture the DJI itself time? (time_sync_callback_sample)

tsai-10957111, 2021-01-15:
How to execute time_sync_callback_sample from DJI M210 v2?

dji-dev, 2021-01-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43076'>Zendesk ticket #43076</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About OSDK time synchronization function, you can visit the website for the related introduction first
https://developer.dji.com/document/264e0359-bbb7-469a-9954-0a9bcdcca878

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

tsai-10957111, 2021-01-18:
Can I write programs in python?

dji-dev, 2021-01-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43076'>Zendesk ticket #43076</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The official lib and demo are based on C/C++, If you need to use other programming languages, you need to perform secondary development based on the official lib.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-25 by dji-kyle