dji-sdk/Onboard-SDK #718: How to solve No such file or directory?

tsai-10957111, 2021-01-21:
[
[image]
](url)

dji-dev, 2021-01-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43264'>Zendesk ticket #43264</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

To use OSDK, you need to include the DJI OSDK library and the corresponding header files. It is recommended to check your application code project environment

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

tsai-10957111, 2021-01-22:
May I add the dji osdk library to the .cpp file of time_sync_poll_sample?

tsai-10957111, 2021-01-22:
How to join the dji osdk library?

Hepan02, 2021-05-05:
Hi tsai.

Add this line to CMakeList.txt in your sample.

```
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../../../core/inc)
include_directories(include /home/pi/Onboard-SDK/sample/platform/linux/common)
include_directories(include /home/pi/Onboard-SDK/sample/platform/linux/hal)
include_directories(include /home/pi/Onboard-SDK/sample/platform/linux/osal)
include_directories(include /usr/local/include)
include_directories(include /usr)
```

closed 2021-06-25 by dji-kyle