dji-sdk/Onboard-SDK #720: 关于OSDK3.9里面支持的航点模式里面航点动作列表中支持的功能的介绍

kimberz-git, 2021-01-24:
我在翻看前面已贴出的问题里面看到，有些回复里面有描述到可以通过设置WayPointSettings的 commandList 和 commandParameter 实现航点动作，其中 commandList 代表航点动作的列表，请问哪里可以查到关于航点动作列表以及相关其他功能的文档描述，我在OSDK3.9的介绍文档里面没有看到相关的描述。

dji-dev, 2021-01-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43328'>Zendesk ticket #43328</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

关于设置WayPointSettings的 commandList 和 commandParameter 实现航点动作，动作定义如下，OSDK3.9中可能缺少该部分定义，该定义已经补充至OSDK 4.0中，对应部分的定义与3.9是一致的，可以参考。
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/NxYb9fFmzIlrHcxndzTs3rAvU' target='zd_attach'>inline762833767.png</a>

kimberz-git, 2021-01-27:
> Agent comment from kyle.cai in [Zendesk ticket #43328](https://djisdksupport.zendesk.com/agent/tickets/43328):
> 
> 尊敬的客户，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 关于设置WayPointSettings的 commandList 和 commandParameter 实现航点动作，动作定义如下，OSDK3.9中可能缺少该部分定义，该定义已经补充至OSDK 4.0中，对应部分的定义与3.9是一致的，可以参考。
> [image]​
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> [inline762833767.png](https://djisdksupport.zendesk.com/attachments/token/NxYb9fFmzIlrHcxndzTs3rAvU)

您好，请问可以给一个地址吗，我搜了下github和https://developer.dji.com/里面的doc，没找到像OSDK 3.9的一样提供了一个好几百兆的文档供我参考使用。
我在这个网站里面的文档描述里面没找到您贴出来的这些
https://developer.dji.com/onboard-sdk/documentation/

dji-kyle, 2021-06-25:
参数说明已经添加到了OSDK 4.x版本的代码注释中，可以参考移至3.9版本代码中。

closed 2021-06-25 by dji-kyle