dji-sdk/Onboard-SDK #721: 4.1.0 Release

lorenzave, 2021-01-26:
On your documentation, you say you would release version 4.1.0 on January 20th but we haven't seen any update. Will you release the updated drone firmware soon? We look forward to the SD card fixes :) 

Thanks.

dji-dev, 2021-01-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43399'>Zendesk ticket #43399</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, OSDK 4.0.1 has not been officially released yet. You can download the OSDK master branch to verify related functions first.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2021-01-27:
[image]
Several READMEs say the opposite :) 
Is there any real release date? Are you releasing everything in master to 4.1.0?
Thanks.

dji-dev, 2021-01-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43399'>Zendesk ticket #43399</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the trouble. OSDK 4.1.0 will have an official branch release. For some reasons, please ignore the readme part. The master branch version only updates the beta version, and the official version 4.1.0 will be released on a scheduled date.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2021-01-27:
You mention a new FW version for the M300, is that needed to check the new released features?

closed 2021-02-25 by lorenzave