dji-sdk/Onboard-SDK #723: 关于Waypoint Mission V1/V2的问题

kimberz-git, 2021-01-31:
您好：
     我在看OSDK3.9的版本的SMT32平台的example的时候看到的waypoint mission的函数看起来像是V1的模式，请问OSDK3.9支持Waypoint Mission V2的功能吗？

dji-dev, 2021-02-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43519'>Zendesk ticket #43519</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

OSDK waypoint v2功能仅支持和适配OSDK 4.0 和M300机型。3.9版本中请使用waypoint v1。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

kimberz-git, 2021-02-09:
> Agent comment from kyle.cai in [Zendesk ticket #43519](https://djisdksupport.zendesk.com/agent/tickets/43519):
> 
> 尊敬的客户，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> OSDK waypoint v2功能仅支持和适配OSDK 4.0 和M300机型。3.9版本中请使用waypoint v1。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持

了解了，非常感谢

closed 2021-02-09 by kimberz-git