dji-sdk/Onboard-SDK #724: vehicle->waypointV2Mission->downloadMission(Mission, timeout)获取数据错误

vickiegem, 2021-01-31:
vehicle->waypointV2Mission->downloadMission(downloadMission, timeout);
使用此接口获取到的路点任务信息与上传的信息不同。

start waypoint mission: waypoint[0]: latitude : 0.394272, longitude : 1.98932, relativeHeight : 30"
start waypoint mission: waypoint[1]: latitude : 0.394263, longitude : 1.98933, relativeHeight : 30"
start waypoint mission: waypoint[2]: latitude : 0.394263, longitude : 1.98933, relativeHeight : 30"
start waypoint mission: waypoint[3]: latitude : 0.394261, longitude : 1.98933, relativeHeight : 30"
start waypoint mission: waypoint[4]: latitude : 0.39426, longitude : 1.98933, relativeHeight : 30"

get waypoint mission: waypoint[0]: latitude : 0.394272, longitude : 1.98932, relativeHeight : 30"
get waypoint mission: waypoint[1]: latitude : 0.39428123, longitude : 1.9893103, relativeHeight : 30"
get waypoint mission: waypoint[2]: latitude : 0.39428123, longitude : 1.9893103, relativeHeight : 30"
get waypoint mission: waypoint[3]: latitude : 0.39428123, longitude : 1.9893081, relativeHeight : 30"
get waypoint mission: waypoint[4]: latitude : 0.39428123, longitude : 1.989307, relativeHeight : 30"

这是上传及下载的数据对比。

希望早点修复此bug。

dji-dev, 2021-02-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43522'>Zendesk ticket #43522</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

请问您使用的固件版本和OSDK版本分别是什么呢？

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

vickiegem, 2021-02-01:
[image]
我找到问题所在了，是这里的XY反了，我将其换过来过，验证结果是正确的

dji-dev, 2021-02-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43522'>Zendesk ticket #43522</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

关于waypoint v2，X/Y反的问题，问题已经修复，可以下载最新的版本进行验证和使用。
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/EaC0ulWJ5r2CCKQtC8uMTIfCF' target='zd_attach'>inline2012183199.png</a>

vickiegem, 2021-02-02:
不，我说的并不是这个问题，249e0cd的提交解决的是上传路点数据错误，我现在发现的问题是下载路点数据错误，请认真看我附图代码的行数。

dji-dev, 2021-02-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43522'>Zendesk ticket #43522</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

​建议您可以先试下，这个与代码行数没有必然关系，关于Waypoint V2 X/Y 反的问题在于上传航点任务时编码上传的X/Y值反了，所以下载路点数据是反的，图示commit修正了上传路点的X/Y值，所以如果当前排查到的问题是X/Y是反的，更新此代码即可修正。若问题依旧存在，建议再提供一下详细操作和复现问题信息，这边将进一步验证和排查。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

vickiegem, 2021-02-02:
ok，下载路点数据有问题，我就是用的目前master代码，这应该是包括上述提到的commit，修复的是上传数据有误，这会造成实际路线与预想路线不一致的情况。目前的master代码中没有路线错误的问题，说明上传的数据是正确的。我现在需要获取到正确的正在执行中的路点数据，但是下载到的数据有误，这应当是下载时数据转换有误，并不是上传数据错误。上面的代码行数是我描述错误，我本意是想让你们关注的是具体接口。

dji-dev, 2021-02-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43522'>Zendesk ticket #43522</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

抱歉前面理解有误，现在已了解您的意思，下载接口中的数据转换查看确实依旧存在错误。感谢您的反馈，该问题我们已经再次提交进行修复，此问题你可先在代码中修正此处的X/Y对应经纬度数据，问题若有纠正确认后，我们将再更新至SDK中。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-kyle, 2021-06-25:
问题已修复

closed 2021-06-25 by dji-kyle