dji-sdk/Onboard-SDK #728: 关于DJI Assistant 2支持的无人机机型

kimberz-git, 2021-02-09:
您好:
     我在贵官网查看DJI Assistant 2的软件的时候，看了下几个DJI Assistant 2的版本， 下面所列出来的可支持的机型都没有M600pro这一款，请问现在官网列出的DJI Assistant 2这几个版本都不支持M600 pro吗？有可支持该版本呢的DJI Assistant 2 版本吗？

dji-dev, 2021-02-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43849'>Zendesk ticket #43849</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

M600 Pro机型使用的DJI Assistant2请前往下述链接下载：
https://www.dji.com/cn/downloads/softwares/assistant-dji-2-for-matrice
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/eutnUanlxaMbztEui5yYJjoFl' target='zd_attach'>inline-435684679.png</a>

kimberz-git, 2021-02-09:
> Agent comment from kyle.cai in [Zendesk ticket #43849](https://djisdksupport.zendesk.com/agent/tickets/43849):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> M600 Pro机型使用的DJI Assistant2请前往下述链接下载：
> https://www.dji.com/cn/downloads/softwares/assistant-dji-2-for-matrice
> [image]​
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> [inline-435684679.png](https://djisdksupport.zendesk.com/attachments/token/eutnUanlxaMbztEui5yYJjoFl)

好的，多谢

closed 2021-02-09 by kimberz-git