dji-sdk/Onboard-SDK #729: Gimbal relative rotation doesn't work

lmar, 2021-02-09:
I'm using the latest master version of the onboard sdk and a Matrice 300 RTK with H20T (all with latest firmwares).

I'm trying to rotate the H20T gimbal using a relative command. For example when I send a yaw value of 2 then I expect the gimbal to rotate 2 degrees further in the yaw axis, regardless of where it is currently pointing to.

For this I'm using the GimbalManager::rotateAsync function. Here the parameter GimbalModule::Rotation should allow this by setting rotationMode to 1:

```cpp
// dji_gimbal_module.hpp:71
  typedef struct Rotation {
    /*! rotation cooradiration
     *  0 = execute angle command based on ground frame as the horizontal frame (NEU)
     *  1 = execute angle command based on current gimbal angle
     */
    uint8_t rotationMode;
    /*! pitch angle in degree, unit : deg
     */
    float pitch;
    /*! roll angle in degree, unit : deg
     */
    float roll;
    /*! yaw angle in degree, unit : deg
     */
    float yaw;
    /*! execution time, unit : second
     */
    double time;
  } Rotation;
```

But instead of interpreting the value as relative, it gets interpreted as an absolute value. Staying with the example, that means that the gimbal will actually rotate to the absolute position of 2 degrees in the yaw axis.

The problem seems to be caused by GimbalModule::rotateAsync which assigns rotationMode to gimbalAngleSetting.coordinate. 

```cpp
// dji_gimbal_module.cpp:139
void GimbalModule::rotateAsync(Rotation rotation,
                               void (*userCB)(ErrorCode::ErrorCodeType retCode,
                                              UserData userData),
                               UserData userData) {
  if (getEnable()) {
    gimbalAngleSetting setting = {0};
    setting.yaw_angle = rotation.yaw * 10;
    setting.pitch_angle = rotation.pitch * 10;
    setting.roll_angle = rotation.roll * 10;
    setting.allowance = 10;  /*!< 0.1 degree allowance */
    setting.reference = 0; /*!< default reference (initial point) */
    setting.coordinate = rotation.rotationMode;
[...]
```

I think rotationMode should actually be assigned to gimbalAngleSetting.reference. I have already tested it and it seems to work: setting gimbalAngleSetting.reference to 1 in GimbalModule::rotateAsync will activate the desired relative rotation.

dji-dev, 2021-02-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43856'>Zendesk ticket #43856</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

We will further verify and confirm this issue. If there is any result, we will synchronize it here in time.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2021-02-23:
We are also having issues with the gimbal control. We tried opening an issue in the ROS GitHub but the proposed fix wouldn't work.
We would like to control the gimbal as it used to be with the old DJI wrapper. Just send an angle with respect to the drone's heading so a yaw command of 0 would just point "forward" and yaw of 90 points left, regardless of the drones heading. It makes control so much harder having to account for the drone heading to just look forward we also have noticed that there is a measurement error between the gimbal compass and the drone compass so not even setting it to the drones heading ensures looking "straight". Also disabling yaw control is not an option as we also want to have the option to control gimbal yaw sometimes. 
What combination of the gimbal_module gimbalAngleSetting would help us get this working? ( the comments explaining the different settings are quite difficult to understand.

dji-dev, 2021-07-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43856'>Zendesk ticket #43856</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue has been fixed and updated to master branch.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/MQP0dnIQYWbrv8210HPPVGlAR' target='zd_attach'>inline-829240500.png</a>
<br><p><em><small>°°°</small></em></p>

closed 2021-07-07 by dji-kyle