dji-sdk/Onboard-SDK #730: WayPointStatusPushData Struct - No function?

YamacTan, 2021-02-11:
I am trying to develop software for Waypoint Mission with Onboard SDK on DJI A3. A struct was written to see if the task ended after it started, but there is no return function for this struct. As a result I cannot use this function. I am sending the struct code block below. 

Is there a point I'm missing? What does this have to do with the ground station protocol? Is there a way to return to this struct and find out that the task is done?

**Thanks.
Y.**

```
/*!
   * @brief This struct captures PushData while ground-station is enabled on
   * Assistant's SDK Page, CMD: 0x02, 0x03
   */
  typedef struct WayPointStatusPushData{
    uint8_t mission_type;      /*! see WayPointPushDataIncidentType */
    uint8_t waypoint_index;    /*! the index of current waypt mission */
    uint8_t current_status;    /*! 0: pre-mission, 1: in-action, 5: first waypt , 6: reached */
    uint8_t error_notification;
    uint16_t reserved_1;
  } WayPointStatusPushData;
  /*!
```

dji-dev, 2021-02-12:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/43893'>Zendesk ticket #43893</a>:<br><br>Hi. The return function is setWaypointCallback. The end of the waypoint mission can be determined by whether the aircraft has reached the last waypoint and finished the waypoint action.

YamacTan, 2021-02-12:
Hi again, I received related datas with creating a callback function I am seeing 4 and 6 for current status like as waypointReachedData. This is the callback that I wrote:

void missioneventCallback (Vehicle* vehicle, RecvContainer recvFrame, UserData userData)
{
    DSTATUS("Wp event callback, \terror_notification: %d \twaypoint_index: %d\tcurrent_status: %d\n,
    recvFrame.recvData.wayPointStatusPushData.error_notification,
    recvFrame.recvData.wayPointStatusPushData.waypoint_index,
    recvFrame.recvData.wayPointStatusPushData.current_status);
}

But the problem is I cannot see 0,1 and 5. 

[image]

Can it be a bug or something?
Thanks for cooperation.
Y.

dji-dev, 2021-02-15:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/43893'>Zendesk ticket #43893</a>:<br><br>Hi. Do you use OSDK 3.9 version?

YamacTan, 2021-02-15:
Hi again, yes it is OSDK 3.9.

dji-dev, 2021-02-18:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/43893'>Zendesk ticket #43893</a>:<br><br>Do you use setWaypointCallback? Generally only use setWaypointEventCallback the current_status will be 4 and 6.

closed 2021-02-19 by YamacTan