dji-sdk/Onboard-SDK #732: FlightAnomaly wrong values

deadman2000, 2021-02-19:
M300 always send abnormally value of aircraftIsFalling = 1

dji-dev, 2021-02-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44041'>Zendesk ticket #44041</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

What is the state of the aircraft when you get abnormal value: aircraftIsFalling = 1? Is it a real flight state or a problem found in the simulator?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

deadman2000, 2021-02-20:
In all states. When engines is off and in flight. In simulation and in real flight

dji-dev, 2021-02-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44041'>Zendesk ticket #44041</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

We have submitted this issue for further confirmation. If there is any result, we will promptly feedback with you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2021-04-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44041'>Zendesk ticket #44041</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The issue is expected to be fixed in June with new firmware（M300）.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2022-03-31 by dji-kyle