dji-sdk/Onboard-SDK #733: om_download_sample

Jessicahjj, 2021-02-20:
m300rtk, osdk4.1, tx2
运行om_download_sample.cpp，程序运行到dji_mop_server.cpp文件中的mop_bind_channel函数时进入了死循环，但函数具体实现无法获知，请问这该怎么解决？谢谢！

Jessicahjj, 2021-02-20:
Read App ID
User Configuration read successfully. 

[3176487.578]STATUS/1 @ getDroneVersion, L1702: ret = 0
[3176487.579]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9800BCKFY
[3176487.579]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.69
[3176487.579]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3176487.581]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3176488.580]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3176488.781]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[3176488.781]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3176488.808]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[3176488.808]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[3176489.809]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[3176489.812]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[3176489.812]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[3176489.813]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3176489.813]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[3176489.816]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3176489.817]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[3176489.819]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[3176489.819]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[3176490.811]STATUS/1 @ firewallTask, L112: firewall task created ...
[3176490.914]STATUS/1 @ getDroneVersion, L1702: ret = 0
[3176490.914]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9800BCKFY
[3176490.914]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.69
[3176490.928]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[3176490.928]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[3176490.930]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[3176490.932]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[3176490.932]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3176490.932]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3176491.732]STATUS/1 @ init, L254: Start advanced sensing initalization
[3176491.732]STATUS/1 @ activate, L1329: version 0x3040845

[3176491.823]STATUS/1 @ activate, L1367: Activation successful

OSDK<--->MSDK MOP Sample!
please choose connection type!
a: reliable connection
b: unreliable connection
a
[3176498.581]STATUS/1 @ main, L716: Server task create success!

[3177490.919]STATUS/1 @ accept, L50: /*! 0.Find whether the pipeline object is existed or not */
[3177490.919]STATUS/1 @ accept, L56: /*! 1.Create handler for binding */
[3177490.919]STATUS/1 @ accept, L64: /*! 2.Do binding */
[3177492.919]STATUS/1 @ accept, L77: /*! 3.Do accepting */
[3177492.920]STATUS/1 @ accept, L78: Do accepting blocking for channel [49155] ...

dji-dev, 2021-02-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44067'>Zendesk ticket #44067</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

MOP功能是SDK互联互通，需要再SDK间先创建通道，该sample进入等待连接状态，需要MSDK端发起连接。关于创建连接的逻辑可以根据需要自行修改和实作。请问您是accept状态下，MSDK无法发起连接吗？

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

closed 2021-06-25 by dji-kyle