dji-sdk/Onboard-SDK #736: Inaccurate yaw readings

WenTheProgrammer, 2021-02-22:
I observed that the outputs from TOPIC_QUATERNION are very unstable and inaccurate for yaw angle (I converted quaternions to Euler angles). Without moving my M600 Pro, the yaw angle could be very different before and after I reboot the drone (let's say a 20 degree difference), and there was one time the yaw angle drifted by itself. On the other hand roll and pitch angles are completely normal. Based on the quaternion outputs, when yaw angle is 0, does the X (roll) axis point towards true north or magnetic north? What are some of the possible reasons for the inaccurate yaw angle measurements? How can I improve the yaw measurement accuracy? Thank you!

dji-dev, 2021-02-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44133'>Zendesk ticket #44133</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

You can first calibrate the M600 through DJI Assistant2 and test it in a place with strong GPS signal. If there are other DJI drones, you can also obtain data to compare whether there are errors.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

WenTheProgrammer, 2021-02-23:
Does a 0 yaw degree represent true north or magnetic north?

dji-dev, 2021-02-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44133'>Zendesk ticket #44133</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

If the drone knows its latitude and longitude information(ever searched for satellites after startup), the magnetic declination information is automatically calculated, and yaw angle's reference  is based on the geographic north.
Otherwise, the satellite has never been searched, the yaw angle reference  will only be geomagnetic north.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-02-23 by WenTheProgrammer