dji-sdk/Onboard-SDK #742: DJI OSDK4.0中，waypointV2怎么知道当前飞到第几个航点?

Chauncy-Guo, 2021-03-13:
DJI OSDK4.0中，waypointV2怎么知道当前飞到第几个航点呀?

看到在waypoint v 1中可以用setWaypointEventCallback 获取到当前飞到了第几个航点，但是waypointv2中用那个函数来实现呢？

dji-dev, 2021-03-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44943'>Zendesk ticket #44943</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

waypoint V2也有提供对应的回调函数，可以参考demo中相关使用：
WaypointV2MissionSample::WaypointV2MissionSample(Vehicle *vehicle):vehiclePtr(vehicle){
  vehiclePtr->waypointV2Mission->RegisterMissionEventCallback(vehicle->waypointV2Mission, updateMissionEvent);
  vehiclePtr->waypointV2Mission->RegisterMissionStateCallback(vehicle->waypointV2Mission, updateMissionState);
}

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

Chauncy-Guo, 2021-04-11:
但是发现一个问题：
按照此回调函数进行获取无人机航点时，当无人机未飞行到第一个航点，Mission为0，当无人机飞行到第一个航点与第二个航点之间时，Mission同样为0，只有执行到第二个航点Mission才为1。

想问下，如何判断飞机当前是否已经经过第一个航点了呢？

closed 2022-03-14 by Chauncy-Guo