dji-sdk/Onboard-SDK #747: M300: Catch pause button pressing on the remote control

ddobrolyubova, 2021-03-23:
Hello, DJI Support

Could you explain how I'm able to catch the pressing of the pause button on the M300 remote control?

Best Regards,
Daria Dobrolyubova
ugcs.com

dji-dev, 2021-03-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45381'>Zendesk ticket #45381</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry that the Onboard-SDK does not support catch pause button pressing on the remote control.
If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

ddobrolyubova, 2021-03-24:
Thank you for your reply.
I've submitted the request.

Best Regards,
Daria Dobrolyubova

closed 2021-03-24 by ddobrolyubova