dji-sdk/Onboard-SDK #748: NEW_DEVICE_ERROR/OSDK lost connection with Drone

github4529, 2021-03-24:
firefly@firefly:~/Onboard-SDK/build/bin$ sudo ./camera-stream-poll-sample 
Rga built version:06fc7c4 
Please enter the type of camera stream you want to view
m: Main Camera
f: FPV  Camera
b: both Cameras
m
Read App ID
User Configuration read successfully. 

[1642217.835]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1642217.835]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8500B2474
[1642217.835]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[1642217.835]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[1642217.836]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[1642218.836]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[1642219.036]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[1642219.036]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[1642220.322]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[1642220.323]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[1642221.325]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[1642221.328]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[1642221.328]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[1642221.331]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1642221.331]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[1642221.333]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1642221.333]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[1642221.335]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[1642221.335]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[1642222.326]STATUS/1 @ firewallTask, L112: firewall task created ...
[1642222.350]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1642222.350]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8500B2474
[1642222.350]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[1642222.387]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[1642222.387]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[1642222.389]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[1642222.389]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[1642222.389]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[1642222.389]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[1642223.190]STATUS/1 @ init, L254: Start advanced sensing initalization
[1642223.190]STATUS/1 @ activate, L1329: version 0x3040848

[1642223.961]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[1642224.215]STATUS/1 @ activate, L1360: Retry to activate again ..
[1642224.989]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[1642225.244]STATUS/1 @ activate, L1360: Retry to activate again ..
[1642226.016]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[1642226.272]STATUS/1 @ activate, L1360: Retry to activate again ..
[1642226.299]STATUS/1 @ getErrorCodeMessage, L744: activate
[1642226.299]STATUS/1 @ getCMDSetActivationMSG, L794: NEW_DEVICE_ERROR

[1642226.299]ERRORLOG/1 @ activate, L1380: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

[1642226.299]ERRORLOG/1 @ activate, L1387: Solutions for NEW_DEVICE_ERROR:
	* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
	* If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
	For different aircraft, the App and the version of DJI Ass
[1642226.300]STATUS/1 @ getErrorCodeMessage, L744: initVehicle
[1642226.300]STATUS/1 @ getCMDSetActivationMSG, L794: NEW_DEVICE_ERROR

[1642226.318]STATUS/1 @ getErrorCodeMessage, L744: verify
[1642226.319]STATUS/1 @ getCMDSetSubscribeMSG, L828: SUBSCRIBER_ILLEGAL_DATA_LENGTH

[1642226.341]STATUS/1 @ getErrorCodeMessage, L744: reset
[1642226.341]STATUS/1 @ getCMDSetSubscribeMSG, L828: SUBSCRIBER_ILLEGAL_DATA_LENGTH

[1642226.763]STATUS/1 @ ~AdvancedSensing, L186: Destroy task for M300's liveview!
[1642226.764]STATUS/1 @ ~AdvancedSensing, L194: Destroy task for M300's advanced sensing!
Segmentation fault

github4529, 2021-03-24:
#484

closed 2021-03-24 by github4529

dji-dev, 2021-03-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45395'>Zendesk ticket #45395</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For the error : NEW_DEVICE_ERROR, you can also activate the application with DJI Assistant2.
1, Connects the drone and DJI Assistant2 (logon with your account/password)
2,run the sample twice with correct AAPID/APPKEY in UserConfig.txt

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

acordobes-airvant, 2022-11-28:
I have the same problem. I have done this multiple times and it still gives the same error

soussanpaul, 2023-05-23:
Hello, I tried running the telemetry sample and activated the app and ran multiple times the sample. I also verified that the APP ID and KEY were good. I get the same error as you. No matter what i do, it still does'nt work. Did you manage to find a solution ?