dji-sdk/Onboard-SDK #749: M300 camera stream poll example

github4529, 2021-03-24:
firefly@firefly:~/Onboard-SDK/build/bin$ sudo ./camera-stream-callback-sample 
Rga built version:06fc7c4 
Please enter the type of camera stream you want to view
m: Main Camera
f: FPV  Camera
m
Read App ID
User Configuration read successfully. 

[1646624.058]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1646624.058]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8500B2474
[1646624.058]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[1646624.058]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[1646624.059]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[1646625.060]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[1646625.259]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[1646625.260]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[1646625.288]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[1646625.288]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[1646626.291]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[1646626.299]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[1646626.299]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[1646626.301]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1646626.301]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[1646626.317]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1646626.317]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[1646626.325]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[1646626.325]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[1646627.293]STATUS/1 @ firewallTask, L112: firewall task created ...
[1646627.308]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1646627.308]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8500B2474
[1646627.309]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[1646627.380]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[1646627.380]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[1646627.384]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[1646627.385]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[1646627.385]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[1646627.385]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[1646628.187]STATUS/1 @ init, L254: Start advanced sensing initalization
[1646628.187]STATUS/1 @ activate, L1329: version 0x3040848

[1646628.207]STATUS/1 @ activate, L1367: Activation successful

mpi: mpp version: 1c28365a author: T-Firefly Firefly first release
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:104
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:154
mpp_device: mpp_device_init failed to find device for coding 7 type 0
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Segmentation fault

dji-dev, 2021-03-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45398'>Zendesk ticket #45398</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The error seems that it not the sample code segment fault, Could you please provide the OSDK version and Onboard computer type ?
and if is convient, please provide the stack trace info with GDB.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

github4529, 2021-03-24:
> Agent comment from kyle.cai in [Zendesk ticket #45398](https://djisdksupport.zendesk.com/agent/tickets/45398):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> The error seems that it not the sample code segment fault, Could you please provide the OSDK version and Onboard computer type ?
> and if is convient, please provide the stack trace info with GDB.
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team

RK3399Pro

Ubuntu18.04

GDB info:

85	  Vehicle*   vehicle = linuxEnvironment.getVehicle();
(gdb) 
86	  const char *acm_dev = linuxEnvironment.getEnvironment()->getDeviceAcm().c_str();
(gdb) 
87	  vehicle->advancedSensing->setAcmDevicePath(acm_dev);
(gdb) 
88	  if (vehicle == NULL)
(gdb) 
94	  bool fpvCamResult = false;
(gdb) 
95	  bool mainCamResult = false;
(gdb) 
97	  if(f)
(gdb) 
106	  if(m)
(gdb) 
108	    mainCamResult = vehicle->advancedSensing->startMainCameraStream();
(gdb) 
mpp[21788]: mpp_info: mpp version: 3fedc6cd author: zouxf         2020-10-23 Merge remote-tracking branch 'firefly-linux/firefly'
mpp[21788]: mpp_rt: NOT found ion allocator
mpp[21788]: mpp_rt: found drm allocator
mpp[21788]: hal_h264d_api: Assertion vcodec_type & ((1 << VPU_CLIENT_RKVDEC) | (1 << VPU_CLIENT_VDPU1) | (1 << VPU_CLIENT_VDPU2)) failed at hal_h264d_init:104
mpp[21788]: hal_h264d_api: hal_h264d_init hard mode error, value=0
mpp[21788]: hal_h264d_api: Assertion 0 failed at hal_h264d_init:154
mpp[21788]: mpp_device: mpp_device_init failed to find device for coding 7 type 0

Thread 1 "camera-stream-p" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in  ()
#1  0x0000007fb5c9d410 in hal_h264d_init () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#2  0x0000007fb5c2dcb4 in mpp_hal_init () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#3  0x0000007fb5c2b6e4 in mpp_dec_init () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#4  0x0000007fb5c2220c in Mpp::init(MppCtxType, MppCodingType) () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#5  0x0000007fb5c2594c in mpp_init () at /usr/lib/aarch64-linux-gnu/librockchip_mpp.so.1
#6  0x0000007fb72cd1f8 in  () at /usr/lib/aarch64-linux-gnu/libavcodec.so.58
(gdb)

github4529, 2021-03-24:
OSDK branch master

dji-dev, 2021-03-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45398'>Zendesk ticket #45398</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

According to stack trace info, segment fault does not occur in the OSDK, Please check if you are using your own decoding code instead of the original demo's method, and check the platform you are using for decoding.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-25 by dji-kyle