dji-sdk/Onboard-SDK #751: Truncated MainCameraStream

Napalys, 2021-03-29:
Dear DJI team,

I am currently facing an issue, where whenever I am invoking startMainCameraStream I am getting: 

[image]

I am currently using the FFmpeg version (I've also tried with different versions of ffmpeg - same luck there): 

[image]

I have also tried running startH264Stream and decoding it manually with the FFmpeg command "ffmpeg -flags2 showall -i main.h264 ouput6.mp4". However, the issue was still present:

```
Input #0, h264, from 'main.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1440, 29.25 fps, 30 tbr, 1200k tbn, 60 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[h264 @ 0x4846c0] SEI type 1 size 904 truncated at 832
[h264 @ 0x4b7ae0] SEI type 1 size 904 truncated at 832
[h264 @ 0x4d2070] SEI type 1 size 904 truncated at 832
[h264 @ 0x60fba0] SEI type 1 size 904 truncated at 832
[h264 @ 0x6151f0] SEI type 1 size 904 truncated at 832
main.h264: corrupt decoded frame in stream 0
[h264 @ 0x8d2dd0] SEI type 1 size 904 truncated at 832
[libx264 @ 0x4a24f0] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x4a24f0] profile High, level 5.0
[libx264 @ 0x4a24f0] 264 - core 148 r2643 5c65704 - H.264/MPEG-4 AVC codec - Copyleft 2003-2015 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=9 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=crf mbtree=1 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'ouput6.mp4':
  Metadata:
    encoder         : Lavf58.45.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 1920x1440, q=-1--1, 30 fps, 15360 tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.91.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: N/A
[h264 @ 0x6151f0] SEI type 1 size 896 truncated at 832
[h264 @ 0x6151f0] no frame!
main.h264: corrupt decoded frame in stream 0
    Last message repeated 4 times
Error while decoding stream #0:0: Invalid data found when processing input
frame=  286 fps= 19 q=-1.0 Lsize=    2114kB time=00:00:09.43 bitrate=1835.9kbits/s dup=0 drop=13 speed=0.615x
video:2110kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.196802%
[libx264 @ 0x4a24f0] frame I:2     Avg QP:15.11  size: 10909
[libx264 @ 0x4a24f0] frame P:101   Avg QP:18.64  size: 11230
[libx264 @ 0x4a24f0] frame B:183   Avg QP:21.03  size:  5485
[libx264 @ 0x4a24f0] consecutive B-frames:  4.5% 25.9% 13.6% 55.9%
[libx264 @ 0x4a24f0] mb I  I16..4: 34.6% 64.5%  0.9%
[libx264 @ 0x4a24f0] mb P  I16..4:  5.3% 10.2%  0.6%  P16..4:  6.8%  1.2%  0.5%  0.0%  0.0%    skip:75.3%
[libx264 @ 0x4a24f0] mb B  I16..4:  0.9%  1.0%  0.1%  B16..8:  9.5%  0.6%  0.1%  direct: 1.8%  skip:86.0%  L0:47.1% L1:51.0% BI: 1.9%
[libx264 @ 0x4a24f0] 8x8 transform intra:61.6% inter:74.9%
[libx264 @ 0x4a24f0] coded y,uvDC,uvAC intra: 23.3% 24.5% 2.0% inter: 2.4% 3.8% 0.0%
[libx264 @ 0x4a24f0] i16 v,h,dc,p: 33% 22% 14% 31%
[libx264 @ 0x4a24f0] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 32% 18% 37%  2%  2%  3%  2%  2%  2%
[libx264 @ 0x4a24f0] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 20% 15% 34%  6%  6%  5%  5%  4%  3%
[libx264 @ 0x4a24f0] i8c dc,h,v,p: 73% 15% 11%  1%
[libx264 @ 0x4a24f0] Weighted P-Frames: Y:0.0% UV:0.0%
[libx264 @ 0x4a24f0] ref P L0: 69.4%  9.0% 15.3%  6.2%
[libx264 @ 0x4a24f0] ref B L0: 86.4% 11.8%  1.8%
[libx264 @ 0x4a24f0] ref B L1: 94.8%  5.2%
[libx264 @ 0x4a24f0] kb/s:1812.47
```

Any kind of help would be appreciated.
Best regards,
Napalys

dji-dev, 2021-03-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45685'>Zendesk ticket #45685</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This is because the strem including the SEI  NAL and decode SEI  failed, and this issue does not affect the use of stream and the frame decode, you can also consider removing SEI NAL before decoding
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/xdEFX8Uc040NzcicwhIeYBZnh' target='zd_attach'>inline-1101066357.png</a>

closed 2021-03-30 by Napalys

nZac, 2021-07-07:
@dji-dev how would one accomplish removing the SEI NAL before decoding?