dji-sdk/Onboard-SDK #752: Zenmuse P1 stream not working

lorenzave, 2021-03-31:
Hi Support!

We are trying to integrate to the P1 to use it in a similar fashion as the H20T but we get no h264stream or frames. I tried using the OSDK samples to rule out any implementation issues in our side. We can perfectly use the H20T like this.

## Setup
- M300 fw: 02.02.0301
- P1 fw: 02.02.03.02
- OSDK: 4.1.0

## Description

We first try running "camera-stream-callback-sample"

```
/camera-stream-callback-sample /home/DjiUserConfig.txt 
Please enter the type of camera stream you want to view
m: Main Camera
f: FPV  Camera
m
Read App ID
User Configuration read successfully. 
[2275043.751]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2275043.751]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = ****************
[2275043.751]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[2275043.751]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2275043.751]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2275044.751]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[2275044.951]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[2275044.951]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2275044.980]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[2275044.980]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[2275045.981]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[2275045.991]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[2275045.991]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[2275045.993]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2275045.993]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[2275045.994]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2275045.994]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[2275045.996]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[2275045.997]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[2275046.981]STATUS/1 @ firewallTask, L112: firewall task created ...
[2275047.014]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2275047.014]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDHAK00C7554
[2275047.014]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[2275047.044]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[2275047.044]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[2275047.047]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[2275047.047]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[2275047.047]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[2275047.047]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[2275047.848]STATUS/1 @ init, L254: Start advanced sensing initalization
[2275047.848]STATUS/1 @ activate, L1329: version 0x3040848
[2275047.863]STATUS/1 @ activate, L1367: Activation successful
STATUS/1All components for decoding initialized ...
STATUS/1User callback thread created successfully!
STATUS/1****** Decoder Callback Thread Start ******
[2275048.900]STATUS/1 @ getCameraList, L218: Request start pushing camera info ack = 0
[2275049.921]STATUS/1 @ getCameraList, L235: Request end pushing camera info ack = 0
[2275049.921]STATUS/1 @ getCameraList, L236: mounted	cam0:0 cam2:0 cam3:0
[2275049.921]STATUS/1 @ getCameraList, L237: type   	cam0:0 cam2:0 cam3:0
[2275049.921]ERRORLOG/1 @ startH264Stream, L459: camera[0] is not mounted
Failed to open selected camera
Decoder Callback Thread: Get image time out
[2275049.943]STATUS/1 @ verify, L244: [2275049.974]STATUS/1 @ reset, L551: Decoder Callback Thread: Get image time out
Reset Subscription Successful.
STATUS/1Decoder Callback Thread Stopped...
[2275050.405]STATUS/1 @ ~AdvancedSensing, L186: Destroy task for M300's liveview!
[2275050.405]STATUS/1 @ ~AdvancedSensing, L194: Destroy task for M300's advanced sensing!
```

We also tried running the "djiosdk-liveview-sample":

```
./djiosdk-liveview-sample /home/DjiUserConfig.txt 
Read App ID
User Configuration read successfully. 
[2275080.184]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2275080.184]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = *****************
[2275080.184]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[2275080.184]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2275080.185]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2275081.184]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[2275081.385]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[2275081.385]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2275081.401]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[2275081.401]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[2275082.402]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[2275082.412]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[2275082.412]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[2275082.415]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2275082.415]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[2275082.418]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2275082.418]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[2275082.419]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[2275082.419]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[2275083.403]STATUS/1 @ firewallTask, L112: firewall task created ...
[2275083.432]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2275083.432]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDHAK00C7554
[2275083.433]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.72
[2275083.463]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[2275083.464]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[2275083.466]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[2275083.467]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[2275083.467]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[2275083.467]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[2275084.267]STATUS/1 @ init, L254: Start advanced sensing initalization
[2275084.267]STATUS/1 @ activate, L1329: version 0x3040848
[2275084.295]STATUS/1 @ activate, L1367: Activation successful
| Available commands:                                            |
| [a] Start getting FPV H264 stream sample                       |
| [b] Start getting main camera H264 stream sample               |
| [c] Start getting vice camera H264 stream sample               |
| [d] Start getting top camera H264 stream sample                |
b
[2275093.320]STATUS/1 @ getCameraList, L218: Request start pushing camera info ack = 0
[2275094.340]STATUS/1 @ getCameraList, L235: Request end pushing camera info ack = 0
[2275094.341]STATUS/1 @ getCameraList, L236: mounted	cam0:0 cam2:0 cam3:0
[2275094.341]STATUS/1 @ getCameraList, L237: type   	cam0:0 cam2:0 cam3:0
[2275094.341]ERRORLOG/1 @ startH264Stream, L459: camera[0] is not mounted
[2275094.341]STATUS/1 @ main, L135: Wait 10 second to record stream
Segmentation fault (core dumped)
```

Does OSDK 4.1.0 support the P1?

I look forward to hearing from you.
Thanks.

dji-dev, 2021-04-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45814'>Zendesk ticket #45814</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry that the OSDK and OSDK ROS currently do not support Zenmuse P1.and it is still under evaluation, you can continue to pay attention to the release and update of related functions.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2021-04-08:
Hi Support,
what do you mean under evaluation? You mean it's in development? Or DJI is maybe thinking about not supporting P1 with the OSDK?
What about the L1? are you planning on exposing the distance measurements of the Lidar?
What about the H20T laser rangefinder distance? Will that be exposed anytime soon? 

Thanks

nZac, 2021-06-14:
@dji-dev I have very similar questions to @lorenzave. Can you provide any indication if the P1 will be formally supported through the OSDK? Additionally the H20T and L1 would be of great interest to us as well.

Thanks for your support!
Nick

dji-kyle, 2021-06-25:
L1 has no plans yet, P1 will be supported in OSDK 5.0.0, and it is expected to be at the end of July

closed 2021-06-25 by dji-kyle

nZac, 2021-08-05:
@dji-kyle any update on P1 support?

NickHAnderson019, 2021-09-03:
Same comment as @nZac, any updates on P1 support?