dji-sdk/Onboard-SDK #754: 最新SDK4编译很烦，怎么配置才不需要乱七八糟的ffmpeg/opencv？？？

addriumruss, 2021-04-06:
最新SDK4编译很烦，怎么配置才不需要乱七八糟的ffmpeg/opencv？？？

dji-dev, 2021-04-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46045'>Zendesk ticket #46045</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

当前OSDK官方demo提供的工程环境与提供的demo有关系，ffmpeg和opnecv依赖主要用于应用端的图像相关处理，若不需要这部分，开发者可以评估自行移除高级视觉相关功能和sample，提供的编译环境也需要根据您移除的功能对应进行修改。当前环境中可能无法直接配置。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

addriumruss, 2021-05-14:
自行移除说起来简单，做起来很麻烦，希望OSDK团队能在CMAKELIST文件中提供相关配置开关，或者给出移除图像相关代码的详细可行步骤，
谢谢！

boythl, 2021-05-25:
能不能介绍下 依赖哪个版本的ffmpeg和opencv
我用当前ffmpeg主干代码编译，会报一堆错误。
`/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp: In member function ‘bool DJICameraStreamDecoder::init()’:
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:70:24: error: ‘avcodec_register_all’ was not declared in this scope
   avcodec_register_all();
                        ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:78:32: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
   pCodec = avcodec_find_decoder(AV_CODEC_ID_H264);
                                ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp: In member function ‘void DJICameraStreamDecoder::decodeBuffer(uint8_t*, int)’:
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:222:68: error: ‘avcodec_decode_video2’ was not declared in this scope
       avcodec_decode_video2(pCodecCtx, pFrameYUV, &gotPicture, &pkt);
                                                                    ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:244:62: error: ‘avpicture_get_size’ was not declared in this scope
           bufSize = avpicture_get_size(AV_PIX_FMT_RGB24, w, h);
                                                              ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:246:27: error: ‘AVPicture’ was not declared in this scope
           avpicture_fill((AVPicture*)pFrameRGB, rgbBuf, AV_PIX_FMT_RGB24, w, h);
                           ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:246:37: error: expected primary-expression before ‘)’ token
           avpicture_fill((AVPicture*)pFrameRGB, rgbBuf, AV_PIX_FMT_RGB24, w, h);
                                     ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:246:79: error: ‘avpicture_fill’ was not declared in this scope
           avpicture_fill((AVPicture*)pFrameRGB, rgbBuf, AV_PIX_FMT_RGB24, w, h);
                                                                               ^
/root/manifold/Onboard-SDK-4.1/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp:264:22: error: ‘av_free_packet’ was not declared in this scope
   av_free_packet(&pkt);
`

closed 2021-06-25 by dji-kyle