dji-sdk/Onboard-SDK #755: 我想要获取IMU的数据，订阅函数按照官方历程进行配置，分别为200和400HZ，但是TOPIC_ANGULAR_RATE_RAW、TOPIC_ACCELERATION_RAW、TOPIC_HARD_SYNC的数据的实际更新频率仅仅为25HZ

Chauncy-Guo, 2021-04-11:
我想要从大疆M300无人机中获取IMU数据，但是实际获取到的数据更新频率过低，导致无法使用，想询问下到底是哪里出了问题。

# 我的Matric配置为：
[image]

# 我的subscribe函数是在官方的Telemetry历程中进行修改的，代码为：

`  // Package 3: Subscribe to TOPIC_ANGULAR_RATE_RAW at freq 200 Hz.
  pkgIndex                   = 3;
  freq                       = 200;
  TopicName topicList200Hz[] = { TOPIC_ANGULAR_RATE_RAW };
  numTopic        = sizeof(topicList200Hz) / sizeof(topicList200Hz[0]);
  enableTimestamp = false;

  pkgStatus = vehicle->subscribe->initPackageFromTopicList(
          pkgIndex, numTopic, topicList200Hz, enableTimestamp, freq);
  if (!(pkgStatus))
  {
      return pkgStatus;
  }
  subscribeStatus = vehicle->subscribe->startPackage(pkgIndex, responseTimeout);
  if (ACK::getError(subscribeStatus) != ACK::SUCCESS)
  {
      ACK::getErrorCodeMessage(subscribeStatus, __func__);
      // Cleanup before return
      vehicle->subscribe->removePackage(pkgIndex, responseTimeout);
      return false;
  }`

`
  // Package 5: Subscribe to TOPIC_ACCELERATION_RAW at freq 400 Hz.
  pkgIndex                   = 5;
  freq                       = 400;
  TopicName topicList400Hz[] = { TOPIC_ACCELERATION_RAW ,TOPIC_HARD_SYNC};
  numTopic        = sizeof(topicList400Hz) / sizeof(topicList400Hz[0]);
  enableTimestamp = false;

  pkgStatus = vehicle->subscribe->initPackageFromTopicList(
          pkgIndex, numTopic, topicList400Hz, enableTimestamp, freq);
  if (!(pkgStatus))
  {
      return pkgStatus;
  }
  subscribeStatus = vehicle->subscribe->startPackage(pkgIndex, responseTimeout);
  if (ACK::getError(subscribeStatus) != ACK::SUCCESS)
  {
      ACK::getErrorCodeMessage(subscribeStatus, __func__);
      // Cleanup before return
      vehicle->subscribe->removePackage(pkgIndex, responseTimeout);
      return false;
  }

`

`  while (elapsedTimeInMs < timeToPrintInMs)
  {

    imu_acceleration  = vehicle->subscribe->getValue<TOPIC_ACCELERATION_RAW>();
    std::cout << "imu_acceleration.x = " << imu_acceleration.x << endl;
    std::cout << "imu_acceleration.y = " << imu_acceleration.y << endl;
    std::cout << "imu_acceleration.z = " << imu_acceleration.z << endl;

    hard_synk = vehicle->subscribe->getValue<TOPIC_HARD_SYNC>();
    std::cout << "hard_synk.a.x == " << hard_synk.a.x <<endl;
    std::cout << "hard_synk.a.y == " << hard_synk.a.y <<endl;
    std::cout << "hard_synk.a.z == " << hard_synk.a.z <<endl;

    std::cout << "-------\n\n";
    usleep(2500);
    elapsedTimeInMs += 5;
}
`

这是我的代码，希望官方解答下，谢谢～

dji-dev, 2021-04-12:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46303'>Zendesk ticket #46303</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

请问数据实际更新频率25HZ是通过什么获取的呢？若方便建议您提供一下对比获取数据频率的值或操作方式，以便进一步为您确认。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

Chauncy-Guo, 2021-04-12:
我将获取到到数据进行print显示，这是我执行用到代码，其中usleep为2500，故print的频率为400HZ，在实际输出的数据中，却发现十分规律的，以16组为单位的重复数据。经过多次测试，证实实际数据的更新频率为400/16=25HZ。（程序运行期间飞机一直在前后左右进行晃动）

imu_acceleration  = vehicle->subscribe->getValue<TOPIC_ACCELERATION_RAW>();
std::cout << "imu_acceleration.x = " << imu_acceleration.x << endl;
std::cout << "imu_acceleration.y = " << imu_acceleration.y << endl;
std::cout << "imu_acceleration.z = " << imu_acceleration.z << endl;

hard_synk = vehicle->subscribe->getValue<TOPIC_HARD_SYNC>();
std::cout << "hard_synk.a.x == " << hard_synk.a.x <<endl;
std::cout << "hard_synk.a.y == " << hard_synk.a.y <<endl;
std::cout << "hard_synk.a.z == " << hard_synk.a.z <<endl;

std::cout << "-------\n\n";
usleep(2500);
elapsedTimeInMs += 5;

dji-dev, 2021-04-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46303'>Zendesk ticket #46303</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

请问您是否方便提供您查看到的规律的数值呢？按照您的测试方式，使用400Hz的打印频率，如果实际数值更新仅25Hz，打印的数值因为更新慢会存在连续的重复值，这边直接在sample中修改代码按照您提供的方式打印，未发现有大量的重复值出现，或者相关规律数值。如果方便，建议您提供一下您测试获取的规律数值。
 [image]

 [image]​

​感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/aY8Hh15eyxKkunnjIrVZgVQtp' target='zd_attach'>inline-2110464666.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/ThS4w093wx8RoTcnOuDAN6aIt' target='zd_attach'>inline509804543.png</a>

closed 2021-06-25 by dji-kyle