dji-sdk/Onboard-SDK #758: Xavier NX with M300 - OSDK

Salazar-Prime, 2021-04-15:
Hi, I get the following error on running the sample:

```
Read App ID
User Configuration read successfully. 

[3593442.304]ERRORLOG/1 @ initVehicle, L222: Failed to initialize ACM Linker channel!
[3593443.675]STATUS/1 @ getDroneVersion, L1702: ret = 2
[3593443.675]ERRORLOG/1 @ getDroneVersion, L1707: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are invert
[3593443.675]STATUS/1 @ functionalSetUp, L282: Shake hand with drone Fail ! Cannot get drone version. (1/20)
[3593443.676]STATUS/1 @ functionalSetUp, L284: Try again after 1 second ......

```
Setup
```
Xavier NX
M300
OSDK 4.1
Connection via USB-C breakout board = UART and USB-ttl
API control active in DJI assistant
Baud rate set to 38400
```
UserConfig file

```
app_id: *
app_key : *
device : /dev/ttyUSB0
baudrate : 38400
acm_port : /dev/ttyACM0
```

dji-dev, 2021-04-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46543'>Zendesk ticket #46543</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、About the serial communication，please check the RX/TX pins are connected correctly, and the baud rate setting of the drone is consistent with the configuration file. The baud rate you set is 38400.
It is recommended to use 230400 and above for Linux, and 921600 and above for ROS.
 [image]

2、​And the log shows the USB communication not works, please connects the dual-A USB cables or it maybe impacts some functions of OSDK(such as advance sensing, camera manager, gimbal manager).

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/yJMGHAcAYveEQH6oIMZPncu3l' target='zd_attach'>inline-1151968422.png</a>

Salazar-Prime, 2021-04-16:
Thanks for the suggestions. I will try them out. 

In the meantime, I had a clarification about dual USB a  cable. As I mentioned, I am interfacing with the M300 using a USB-C breakout board. I connect the USB-ttl cable to A6 and A7 pin while the other end is plugged into Xavier NX. 

It is not clear where will I plug in the suggested dual USB A cable.

dji-dev, 2021-04-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46543'>Zendesk ticket #46543</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The type C interface on M300 needs to be expanded to TTL pin and USB interface for serial communication and USB communication respectively.
You can use the official DJI OSDK expansion adapter board, or you can customize the expansion adapter board according to the official type C pin definition.

For device connection or type C pin definition, please refer to the developer's official website：
https://developer.dji.com/document/1bb1b2dc-9694-4d6b-ad43-fa69c5a10400

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-25 by dji-kyle