dji-sdk/Onboard-SDK #760: UWB localization data & vehicle control

gerhardmalberg88, 2021-04-20:
Hello,

we are using DJI M100 without RTK. 
OSDK 3.9.
Our mission is to use Machine Vision and UWB sensor fused localization data which will replace GPS data. GPS use is avoided since we are developing an indoor system. GPS is connected but data not utilized. 
We have been testing out what commands does our onboard computer gives to the vehicle. 
moveByPositionOffset has been modified so that our data is replaced with original broadcasted data from GPS and seems that our onboard computer forwards the commands to FC but the vehicle seems not to control anything. 
positionAndYawCtrl function does not control anything, not even Yaw, however, we tried another function attitudeAndVertPosCtrl which works perfectly regarding our commands.

I am assuming now that since we are doing indoor flying, there is no sufficient GPS data (4 satellites) thereby drone will not control using the commands from the onboard computer even if they equal the same? Something tied to the firmware?

Our goal is to avoid making our own control model by using the attitudeAndVertPosCtrl for example.
Question is that are there any functions or parameters to be set that can give us control and pass this GPS check and to be able to utilize the moveByPositionOffset (what we assume)?

And what would be the correct way to run this type of localization system on your vehicle?

Best Regards,
Malberg & Team

dji-dev, 2021-04-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46762'>Zendesk ticket #46762</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Because the current DJI drones and OSDK do not provide a complete non-GPS environment control solution, only the OSDK attitude control API is recommended for indoor environments without GPS.

moveByPositionOffset is also available, but the actual command sent to FC is the following API:
https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/sample/platform/linux/flight-control/flight_control_sample.cpp#L461

The moveByPositionOffset can be combined with external sensors to implement closed-loop control.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

gerhardmalberg88, 2021-04-21:
Hello
Thank You for your answer

1. "The moveByPositionOffset can be combined with external sensors to implement closed-loop control."
Am I understanding correctly that this API:
vehicle->control->positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD);
still CAN be used for indoor non-GPS with our provided localization data closed-loop control? 
Or there is no way the flight controller will let us use this method indoors even since we get reasonable localization data?

2. Is there any possibility to reflash FC with any "to be able to do that" firmware solution? Any hacks?  (We are using this for academic purpose)

Best Regards,
Gerhard Malberg

gerhardmalberg88, 2021-04-21:
I see that a GPS health flag "health_flag" is restricting the use of control flag HORIZONTAL_POSITION thereby the use of positionAndYawCtrl API.

Is there any way to set this health_flag manually healthy all the time to bypass the GPS check? 
where is this located in your library?

Best regards,
Malberg

dji-dev, 2021-04-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46762'>Zendesk ticket #46762</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、"The moveByPositionOffset can be combined with external sensors to implement closed-loop control."
Am I understanding correctly that this API:
vehicle->control->positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD);
still CAN be used for indoor non-GPS with our provided localization data closed-loop control?
    -->Yes.

2、Is there any possibility to reflash FC with any "to be able to do that" firmware solution? Any hacks? (We are using this for academic purpose)
   -->Sorry, it is not supported.

3、I see that a GPS health flag "health_flag" is restricting the use of control flag HORIZONTAL_POSITION thereby the use of positionAndYawCtrl API.
Is there any way to set this health_flag manually healthy all the time to bypass the GPS check?
where is this located in your library?
   -->Non-GPS environment does not need to obtain GPS signal status from the flight controller, "flag" can be implemented the code according to your needs, not using the GPS health_flag from the FC.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

gerhardmalberg88, 2021-04-22:
Thank You, this was clarifying answer.

With default control flags the vehicle is doing nothing and only starts to drift a little in random direction when API "vehicle->control->positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD)" is used. It does not even control Yaw.
We confirm that: 
*correct "xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD" values are calculated and passed to the API
*obtainControl has been done (tested and works with other API's)
*other supported API works (attitudeAndVertPosCtrl)
*remote set to (Function) Mode and Enable Multiple Flight Mode ON

1. What are the correct control flags for API "positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD)" since we have our own localization data?

2. What are other possible factors for that "positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD)" API does not work in our case?
Something need to be configured in DJI assistant?
Remote settings?
Other flags?

Best Regards,
Malberg

dji-dev, 2021-04-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46762'>Zendesk ticket #46762</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

With default control flags the vehicle is doing nothing and only starts to drift a little in random direction when API "vehicle->control->positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD)" is used. It does not even control Yaw.
    -->Yes, this is for the safety of OSDK's flight control. OSDK needs to call positionAndYawCtrl at a certain frequency. Once it stops calling the API, the drone will stop flighting.

This can also explain that the actual position control closed loop is implemented in the function moveByPositionOffset. The parameter of positionAndYawCtrl is only used as the effective parameter of the control command, not the actual position parameter.

 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/7LaJMYNne3Qmlgc7DmHxDC1rz' target='zd_attach'>inline909910964.png</a>

gerhardmalberg88, 2021-04-27:
We decided to use the API attitudeAndVertPosCtrl and create our own controller.

Thank You,
Best Regards,
Malberg

closed 2021-04-27 by gerhardmalberg88