dji-sdk/Onboard-SDK #764: payloads_download_sample fails to download when highres grid is present on SD card

lorenzmhe, 2021-04-28:
The payloads_download_sample fails to read the file list and download files when a highres grid is present on the SD card.

The test uses the payloads_download_sample with line 105 (` if (downloadCnt > 4) downloadCnt = 4`) commented out to download all files.

The test included the following steps:
0. The SD card was formatted before the test
1. Download main camera file list (no files)
2. Capture two images (WIDE, ZOOM, and THRM) with the smart controller
3. Download main camera file list (6 files)
4. Download 6 images
5. Capture a highres grid with the smart controller
6. Download main camera file list (failed: timeout)
7. Download images (failed)
8. Delete highres grid on with the smart controller
9. Download main camera file list (6 files)
10. Download 6 images

Please see the attached log for results: [payload_download_sample_highres_grid_error_1.txt](https://github.com/dji-sdk/Onboard-SDK/files/6391446/payload_download_sample_highres_grid_error_1.txt).

dji-dev, 2021-04-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47112'>Zendesk ticket #47112</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

We have reproduced this problem and submitted it to relevant engineers for further confirmation and analysis.
Thank you for your feedback. If there is any result, we will communicate with you in time.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-05-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47112'>Zendesk ticket #47112</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About your issue：
payloads_download_sample fails to download when highres grid is present on SD card

We have received updates from relevant departments to synchronize with you as follows：
Sorry, the current OSDK does not support download highres grid of SD card.
If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

lorenzave, 2021-05-10:
We know it's not supported, the issue is that every SD card function fails when there is a high-res grid. It's fine it doesn't download, it would be optimal if it didn't break the SD card flow.

lorenzmhe, 2021-06-16:
Sorry for the late response here. We don't know how we can get more specific than the steps and log in the original issue. As my colleague explains, the problem is that any SD-card function fails when a High-Res grid is present on the SD-card. The specific anomaly is described in the original issue.

closed 2021-07-07 by dji-kyle