dji-sdk/Onboard-SDK #766: Shake hand with drone failed with M300 RTK

wang-zi-tao, 2021-05-01:
# OSDK 运行 demo `./stereo-vision-single-thread-sample` 时报错  :
```
Read App ID
User Configuration read successfully. 

[633673.437]ERRORLOG/1 @ initVehicle, L222: Failed to initialize ACM Linker channel!
[633674.781]STATUS/1 @ getDroneVersion, L1701: ret = 2
[633674.781]ERRORLOG/1 @ getDroneVersion, L1706: Drone version not obtained! Please do not proceed.
Possible reasons:
        Serial port connection:
                * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
                * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
                * TX and RX pins are invert
[635871.319]STATUS/1 @ functionalSetUp, L282: Shake hand with drone Fail ! Cannot get drone version. (1/20)
[635871.319]STATUS/1 @ functionalSetUp, L284: Try again after 1 second ......
```
### 运行demo后，`stty -F /dev/ttyUSB0 -a`:

```
speed 230400 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 0; time = 0;
-parenb -parodd -cmspar cs8 hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon iexten -echo -echoe echok -echonl -noflsh -xcase -tostop -echoprt echoctl echoke -flusho -extproc
```
### 配置文件`UserConfig.txt`:
```
app_id : *
app_key : *
device : /dev/ttyUSB0
baudrate : 230400
acm_port : /dev/ttyACM0
```
USB2TTL模块已经通过回环测试确认支持此波特率
### `cat /etc/udev/rules.d/DJIDevice.rules`:
```
SUBSYSTEM==“usb”, ATTRS{idVendor}==“2ca3”, MODE=“0666”
```
### `lsusb |grep '2ca3:001f'`:
```
Bus 002 Device 004: ID 2ca3:001f
```
### `lsusb --tree` :
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 2: Dev 4, If 0, Class=Wireless, Driver=rndis_host, 5000M
        |__ Port 2: Dev 4, If 1, Class=CDC Data, Driver=rndis_host, 5000M
        |__ Port 2: Dev 4, If 2, Class=Mass Storage, Driver=usb-storage, 5000M
        |__ Port 2: Dev 4, If 3, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 2: Dev 4, If 4, Class=Communications, Driver=cdc_acm, 5000M
        |__ Port 2: Dev 4, If 5, Class=CDC Data, Driver=cdc_acm, 5000M
```
已经开启DJI Assistant 2 中的`启用API控制`选项，并在DJI Assistant 2 中把波特率设置为230400
从`DJI Assistant 2`中查到的固件版本为`V01.00.0211`
## 连接方式：
```
M300 RTK --------------------OSDK EXPANSION MODULE------杜邦线----USB2TTL-----------JETSON NANO
         |                       |                                                    |  |
         |                      +------------------------双A USB线-------------------- +  |
        +----------------------------------Type c线--------------------------------------|
```
    只有在连接上TypeC线后linux系统才识别到`2ca3:001f`usb设备,才会有ip为`192.168.42.1`的网卡接口和`/dev/ACM0`

dji-dev, 2021-05-01:
Agent comment from cheney.zeng in <a href='https://djisdksupport.zendesk.com/agent/tickets/47207'>Zendesk ticket #47207</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

由于五一劳动节放假，将在节后回复您的问题。

感谢您的理解与支持，祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-05-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47207'>Zendesk ticket #47207</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

您的报错显示串口和USB通信均异常，建议先检查硬件连接和配置。
 [image]
上图中的type C线不要连接到机载计算机。​先确认OSDK拓展转接板连接M300的OSDK接口，将拓展 板的ttl通过USB ttl连接至机载计算机，双A-USB连接拓展板的USB和机载端的USB。

建议将飞机固件升级到最新版本，OSDK使用4.1版本。

感谢您的理解与支持，祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/VYZy5IiyBnpELWRn9K493l20M' target='zd_attach'>inline-1674302709.png</a><br><p><em><small>°°°</small></em></p>

closed 2021-06-25 by dji-kyle