dji-sdk/Onboard-SDK #767: Installation problem on Raspberry Pi 4 Debian 10

Hepan02, 2021-05-02:
I can't install Onboard-SDK on my raspberry PI. What I should do?
I used instruction from and was blocked during use **cmake ..**: 
https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html
______________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ **lsb_release -a**
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

______________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ **gcc -v**
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)

______________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ **cmake -version**
cmake version 3.16.3

______________________________________________________________________________________________________
When I use **cmake** i've got an errors:
pi@raspberrypi:~/Onboard-SDK/build $ cmake ..
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
Cannot Find FFMPEG
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
/home/pi/Onboard-SDK/osdk-core/advanced-sensing/api/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/platform/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/protocol/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/camera_stream/src/home/pi/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/protocol/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/hal/inc/home/pi/Onboard-SDK/osdk-core/advanced-sensing/ori-osdk-core/platform/linux/inc
Cannot Find FFMPEG
-----------OSDK_LIB_HEADERS_DIR------------
(/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/api/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/payload/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/flight/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/mop/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/platform/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/logger/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/utility/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/linker/armv7/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/modules/inc/filemgr)
-----------ADVANCED_SENSING_HEADERS_DIR------------
(/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/api/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/platform/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/protocol/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/camera_stream/src/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/ori-osdk-core/protocol/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/ori-osdk-core/hal/inc/home/pi/Onboard-SDK/sample/platform/linux/../../../osdk-core/advanced-sensing/ori-osdk-core/platform/linux/inc)
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
Cannot Find FFMPEG

-- Enable OSDK Hotplug monitoring.
-- Found libudev:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libudev.so

-- You can cmake with "-DCONFIG_TOOL=ON" to rebuild the config tool.

(........)
-- Configuring incomplete, errors occurred!
See also "/home/pi/Onboard-SDK/build/CMakeFiles/CMakeOutput.log".

______________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ cat /home/pi/Onboard-SDK/build/CMakeFiles/CMakeOutput.log
The system is: Linux - 5.10.27-v7l+ - armv7l
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc
Build flags:
Id flags:

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/pi/Onboard-SDK/build/CMakeFiles/3.16.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++
Build flags:
[... 377 lines cut ...]
    arg [/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/crti.o] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/8/crtbegin.o] ==> ignore
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/8] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/8]
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf]
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/8/../../..] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../..]
    arg [-L/lib/arm-linux-gnueabihf] ==> dir [/lib/arm-linux-gnueabihf]
    arg [-L/usr/lib/arm-linux-gnueabihf] ==> dir [/usr/lib/arm-linux-gnueabihf]
    arg [CMakeFiles/cmTC_f2b75.dir/CMakeCXXCompilerABI.cpp.o] ==> ignore
    arg [-lstdc++] ==> lib [stdc++]
    arg [-lm] ==> lib [m]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [/usr/lib/gcc/arm-linux-gnueabihf/8/crtend.o] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/crtn.o] ==> ignore
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/8] ==> [/usr/lib/gcc/arm-linux-gnueabihf/8]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../..] ==> [/usr/lib]
  collapse library dir [/lib/arm-linux-gnueabihf] ==> [/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
  implicit dirs: [/usr/lib/gcc/arm-linux-gnueabihf/8;/usr/lib/arm-linux-gnueabihf;/usr/lib;/lib/arm-linux-gnueabihf]
  implicit fwks: []

Hepan02, 2021-05-02:
The command:
sudo apt install libavcodec-dev libavformat-dev libavfilter-dev

Resolved the problem

dji-dev, 2021-05-03:
Agent comment from cheney.zeng in <a href='https://djisdksupport.zendesk.com/agent/tickets/47222'>Zendesk ticket #47222</a>:<br><br>Hi Developer,

Thank you for reach out to DJI Developer Support.  Due to a Chinese Holiday there maybe a short delay in responding directly to your question - we apologize for any inconvenience this may cause you and look forward to addressing your ticket as soon as possible.

Warm Regards,
DJI Developer Support<br><p><em><small>°°°</small></em></p>

closed 2021-05-07 by Hepan02